microscope-cockpit/cockpit #308: sort out license for cockpit

carandraug, 2018-06-18:
We got permission from UCSF to use cockpit under the 3 clause BSD license.  Our own modifications are under GPLv3+.  At the moment we got both licenses at the root of the project but we need to put the right license headers in the appropriate files.  I went through the history and prepared a commit to do this.

Here's the list of current files in cockpit each of us has modified.  I have placed your names in those files as copyright holders.  Let me know if you think you should be added to any other file, or you want your name or email in some other form:

* @tomparks 
  * `experiment/sweptShutter.py`
  * `gui/loggingWindow.py`
  * `handlers/stagePositioner.py`
* @langefj 
  * `gui/mosaic/window.py`
* @NickHallONBI 
  * `devices/windowsUSBJoystick.py`
  * `interfaces/stageMover.py`
* @juliomateoslangerak 
  * `cockpit.pyw`
  * `devices/camera.py`
  * `devices/ni_cRIOFPGA.py`
  * `experiment/experiment.py`
  * `gui/macroStage/macroStageZ.py`
  * `handlers/camera.py`
  * `util/connection.py`
* @iandobbie 
  * `cockpit.pyw`
  * `devices/aerotechAxis.py`
  * `devices/andorCameras.py`
  * `devices/boulderSLM.py`
  * `devices/configurator.py`
  * `devices/nanomover.py`
  * `devices/ni6036e.py`
  * `devices/objective.py`
  * `devices/physikInstrumenteM687.py`
  * `devices/picomotor.py`
  * `devices/piDIO.py`
  * `devices/polarizationRotator.py`
  * `devices/softCamera.py`
  * `devices/windowsUSBJoystick.py`
  * `experiment/dataSaver.py`
  * `experiment/optoScriptExample.py`
  * `experiment/structuredIllumination.py`
  * `experiment/stutteredZStack.py`
  * `gui/adminWindow.py`
  * `gui/camera/viewPanel.py`
  * `gui/device.py`
  * `gui/dialogs/enumerateSitesPanel.py`
  * `gui/dialogs/experiment/experimentConfigPanel.py`
  * `gui/dialogs/experiment/multiSiteExperiment.py`
  * `gui/dialogs/experiment/singleSiteExperiment.py`
  * `gui/dialogs/getNumberDialog.py`
  * `gui/dialogs/offsetSitesDialog.py`
  * `gui/dialogs/safetyMinDialog.py`
  * `gui/drawShapes.py`
  * `gui/guiUtils.py`
  * `gui/imageViewer/viewCanvas.py`
  * `gui/keyboard.py`
  * `gui/loggingWindow.py`
  * `gui/macroStage/macroStageBase.py`
  * `gui/macroStage/macroStageWindow.py`
  * `gui/macroStage/macroStageXY.py`
  * `gui/macroStage/macroStageZ.py`
  * `gui/mainWindow.py`
  * `gui/mosaic/canvas.py`
  * `gui/mosaic/window.py`
  * `gui/saveTopBottomPanel.py`
  * `gui/toggleButton.py`
  * `gui/touchscreen/slavecanvas.py`
  * `gui/touchscreen/slaveMacroStageZ.py`
  * `gui/touchscreen/slaveOverview.py`
  * `gui/touchscreen/touchscreen.py`
  * `gui/valueLogger.py`
  * `handlers/lightPower.py`
[... 201 lines cut ...]
tom="Thomas Park <thomasparks@outlook.com>"

for AUTHOR in tom rick nick david julio ian mick ; do
  printf "## Copyright (C) 2018 %s\n" "${!AUTHOR}" > copyright-line
  while read FPATH; do
    cat copyright-line "$FPATH" > "$FPATH.tmp"
    mv "$FPATH.tmp" "$FPATH"
  done < $AUTHOR-files
done
```

Put the shebang line and coding lines back in place:

```
for FPATH in $shebang_files $shebang_coding_files ; do
  cat <(printf '#!/usr/bin/env python\n# -*- coding: utf-8 -*-\n\n') $FPATH > $FPATH.tmp
  mv $FPATH.tmp $FPATH
done
```

## Cleanup

* removed the BSD license at the root of the project.  The project is GPL, we only have some files that are also under BSD and those files have the notice as required.
* manually cleaned some files that already had a license header, including Micks Apache 2
* some files only had a line specifying utf-8 coding

carandraug, 2018-06-18:
See commit bab198f for the changes. I had to open the issue first so I could refer to it on the commit message.

iandobbie, 2018-06-20:
List of files sounds reasonable and of course I am happy with the licensing as we have been discussing.

juliomateoslangerak, 2018-06-27:
Hi, I'm happy with the list and the licensing.

J.

iandobbie, 2018-07-13:
Is this not now all done and able to be closed?

carandraug, 2018-07-13:
> Is this not now all done and able to be closed?

Kind of. Not really necessary, but would be nice if the people involved would thumbs up this. It's missing @tomparks, @langefj, @NickHallONBI and @mickp

mickp, 2018-07-13:
Sorry - we discussed this over e-mail and in person, but I never posted anything here. 
👍

tomparks, 2018-07-13:
lgtm, thumbs up from me.

Thanks,
Tom

NickHallONBI, 2018-07-16:
Hi 

Sorry, I had discussed this with a number of people in person but never officially posted here. Yes, I'm happy that those files are the ones I've worked on and with the form of my name and email.

Cheers

Nick

langefj, 2018-07-16:
Hi,

Happy with all as is.

Cheers,
Rick

iandobbie, 2018-07-16:
Ok, I think that is everybody. Can we c lose this issue now?

closed 2018-07-16 by carandraug