microscope-cockpit/cockpit #309: character encoding in deg C symbol.

iandobbie, 2018-06-19:
Just pulled the latest master into cryosim and got text string encoding errors on the deg symbol used in the camera temp and the linkam stage temp. 

eg 
Importing linkamStage failed with Non-ASCII character '\xc2' in file c:\b24\cockpit\devices\linkamStage.py on line 285, but no encoding declared; see http://www.python.org/peps/pep-0263.html for details (linkamStage.py, line 285)

I guess we need to specify an encoding on the string?

mickp, 2018-06-19:
Oof. This will be a Python 3 default-is-unicode vs Python 2
default-is-string problem. I've hit this elsewhere ... it's a massive
Python 3 pain for file IO. I guess this is happening when we scan the file
to look for device classes, before we decide whether or not to import it,
so it's the encoding on the whole file, not one string. I don't know if
specifying an encoding at the top of the file would fix it ... I suspect
not. We probably need to make the file scanner look for an encoding string,
or try...except on each line and failover to Unicode on ASCII errors, or
wrap it in some string buffer class that does this for us?

carandraug, 2018-06-19:
No.  This issue is about declaring the source file itself as being utf-8 encoded. Such declaration needs to be the first or second line and I messed it up when I added the license.  Files that had a shebang line and a encoding declaration are still correct but files that had a encoding declaration alone had them removed.

Should be fixed now, Ian.

carandraug, 2018-06-22:
Ian reported by email that this is now fixed.

closed 2018-06-22 by carandraug