microscope-cockpit/cockpit #311: Problem with FTGL on Windows 10

NickHallONBI, 2018-06-25:
I followed the installation instructions on my Dell XPS 15 running 64 bit Windows 10 and encountered an issue with FTGL. On running "python cockpit.pyw" I get the following error:

"An error occurred during initialization:

[Error 193] %1 is not a valid Win32 application

A full code traceback follows:

Traceback (most recent call last):
File "cockpit.pyw", line 62 OnInit
  import gui.camera.window
File "C:\cockpit\gui\camera\window.py", line 6, in <module>
  import gui.keyboard
File "C:\cockpit\gui\keyboard.py", line 6, in <module>
  import gui.mosaic.window
File "C:\cockpit\gui\mosaic\window", line 2, in <module>
  from util import ftgl
File "C:\cockpit\util\ftgl.py", line 7, in <module>
  _ftgl = ctypes.WinDLL("ftgl")
File "C:\WinPython-64bit-2.7.13.1Zero\python-2.7.13.amd64\lib\ctypes\__init__.py", line 362, in __init__
  self._handle = dlopen(self._name, mode)
[Error 193] %1 is not a valid Win32 application"

I'm not sure at this stage if this is a problem with my specific machine or the installation protocol in general, but the problem appears to be with the pyFTGL DLL grabbed from micronadmin:/cockpitdependencies.

mickp, 2018-06-25:
Looks like you're using a 32-bit library with 64-bit python, maybe?

NickHallONBI, 2018-06-25:
If you're referring to the Win32 part of the error message, I thought that at first.  However, it turns out that Win32 is in fact the 64-bit library and it's only called Win32 because engineers at the time hard-coded the name Win32, so the name stuck.

mickp, 2018-06-25:
Is the file you grabbed pyFTGL.dll, pyFTGL.pyd, FTGL.dll or FTGL.pyd? It
should be FTGL.dll. There used to be a dependency on a python-wrapper for
FTGL built with boost, but I wrapped the FTGL library with ctypes instead,
so you should just need the stock FTGL.dll, not some pyd wrapper. It's
likely that the installation instructions need updating.

NickHallONBI, 2018-06-25:
Yep, it's just the stock FTGL.dll that I grabbed

NickHallONBI, 2018-06-26:
Managed to fix this issue. The FTGL DLL should be put into "C:\WinPython-64bit-2.7.13.1Zero\python-2.7.13.amd64" not "C:\WinPython-64bit-2.7.13.1Zero\python-2.7.13.amd64\Lib". That seemed to fix my issue. I will amend the documentation accordingly.

closed 2018-06-26 by NickHallONBI

mickp, 2018-06-26:
Or just anywhere on the Windows path.