microscope-cockpit/cockpit #314: Running Cockpit on MacOSX

juliomateoslangerak, 2018-06-29:
Usint setup tools python3 setup.py install requires sphinx and pillow.

Running cockpit afterwards in the mac throws:
cockpit/util/ftgl.py", line 31, in <module>
    _ftgl = ctypes.CDLL("libftgl.so") 

You have to add a case to capture the macOS and replace that line by:
_ftgl = ctypes.CDLL("libftgl.dylib")

carandraug, 2018-06-29:
> Usint setup tools python3 setup.py install requires sphinx and pillow.

This applies to microscope, not to cockpit. I have opened an issue there for it.

juliomateoslangerak, 2018-06-29:
You got to use platform and not os:

>>> import platform
>>> platform.system()
'Linux'
>>> platform.release()
'2.6.22-15-generic'
The output of platform.system() is as follows:

Linux: Linux
Mac: Darwin
Windows: Windows

J

carandraug, 2018-07-02:
I have a fix for this ready. I will push it after the setuptools changes to avoid conflicts.

I ended up using sys.platform instead because that's what's used internally by ctypes (for example in `ctypes.util.find_library`).

carandraug, 2018-07-03:
Fix has now been merged into master.

closed 2018-07-03 by carandraug