microscope-cockpit/cockpit #316: Set safeties

juliomateoslangerak, 2018-07-02:
Pressing set safeties raises:
Traceback (most recent call last):
  File "/Users/julio/PycharmProjects/cockpit/cockpit/gui/macroStage/macroStageWindow.py", line 164, in <lambda>
    lambda event: cockpit.gui.dialogs.safetyMinDialog.showDialog(self.GetParent()))
  File "/Users/julio/PycharmProjects/cockpit/cockpit/gui/dialogs/safetyMinDialog.py", line 156, in showDialog
    dialog = SafetyMinDialog(parent)
  File "/Users/julio/PycharmProjects/cockpit/cockpit/gui/dialogs/safetyMinDialog.py", line 126, in __init__
    self.Bind(wx.EVT_BUTTON, wx.ID_OK, self.OnStart)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/wx/core.py", line 1350, in _EvtHandler_Bind
    assert callable(handler) or handler is None
AssertionError

iandobbie, 2018-07-02:
There has always been a weird bug with the stage safety’s. Never been a major problem before so I have so far ignored it.

Ian

iandobbie, 2018-08-28:
Mick's macrostage fixes seem to have fixed the safety loading issue. It all seems to work as expected on DeepSIM so I hope it also works on Julio's setup with these fixes.

mickp, 2018-09-04:
Calling this fixed as of #357 .

closed 2018-09-04 by mickp