microscope-cockpit/cockpit #317: MacOS Quitting does not work

juliomateoslangerak, 2018-07-02:
When quitting cockpit (closing the main window) does not close a the python launcher, which stays as 'Application not responding'.

Maybe a previous error is related...
2018-07-02 15:42:35.640 Python[2002:319550] pid(2002)/euid(501) is calling TIS/TSM in non-main thread environment, ERROR : This is NOT allowed. Please call TIS/TSM in main thread!!!

iandobbie, 2018-07-12:
I don't get the above error but a whole load along the lines...
2018-07-12 17:32:20,128 ERROR      __init__: 263  Destroying <cockpit.gui.mainWindow.MainWindow object at 0x11b249798>

iandobbie, 2018-07-24:
This has been fixed by one of Micks Linux fixes, not sure exactly which one, but now working

closed 2018-07-24 by iandobbie