microscope-cockpit/cockpit #319: Setuptools install missed PIL.

iandobbie, 2018-07-02:
Hi I just grabbed the setuptools branch and when run the first error it generates for me is missing PIL.

carandraug, 2018-07-02:
But PIL is not a dependency of cockpit. I don't see Pil or Pillow being used anywhere in the code. Where does that missing PIL comes from?

iandobbie, 2018-07-02:
Its the shaped button instances, which I guess are in the touchscreen component. And yes what I really needed was pillow.

carandraug, 2018-07-02:
I see it now.  `cockpit.gui.touchscreen.touchscreen` imports `wx.lib.agw.shapedbutton` which imports `PIL`.  However, that's an indirect dependency, we are actually only dependent on Pillow. I'm inclined to fix that on wxPython unless they are doing it on purpose. I will ask them.

carandraug, 2018-07-02:
Asked on IRC, got an answer from its developers:

> carandraug: wx.lib.agw.shapedbutton needs PIL but the setup.py only lists pypubsub and six as the phoenix dependencies. Is that by design?                                                   │
> RobinD: No, it's an oversight. Please create an issue at github about it.

Already submitted wxWidgets/Phoenix#908 so closing this as "won't fix"

carandraug, 2018-11-19:
Because wxPython is not making a new release, we are listing Pillow as requirement. This should be removed once wxPython 4.0.4 is released.

mickp, 2018-12-13:
Somehow, ```Pillow``` became ```Pyllow``` in setup.py. Is this correct? Pyllow looks like somethign to do with Pylons, not PIL.
```
'Pyllow', # temporary until wxPython 4.0.4 release (issue #319)
```

mickp, 2018-12-13:
Yep - looks like this was a typo. Fixed in d0d06d9964fa212f63b70664bfbd46a71d0f927d.

closed 2018-12-13 by mickp

carandraug, 2018-12-13:
Your commit did not make any change on the file, it only changed its permissions.  The typo on the package name had been fixed back in 21e11692c836