microscope-cockpit/cockpit #320: Directories used.

iandobbie, 2018-07-02:
We NEED to sort this out. Sending both the data and the USER directory into a users home dir is WRONG. Some of the system config goes into the config directory and the data almost certainly doesn't want to be in peoples home dir.  These directory changes break lots of little but important things.

carandraug, 2018-07-02:
Where should the data go then?

This has _always_ been the default, see https://github.com/MicronOxford/cockpit/blob/master/util/files.py#L75

I'm guessing the change you are seeing is on Mac.  That's because `devices.configurator` was then using the windows default again, even on Mac.

iandobbie, 2018-07-12:
After discussion on Tuesday we decided....

DataDIR can be specified in standard HW config file so is machine specific.
User config should be in users home directories but have some system id so multipole can coexist.
Global config should be in /etc (linux/mac) or %SystemRoot%/programData (windows)

iandobbie, 2018-07-13:
Further discussion today makes me think that the hardware config file (eg dome.conf) should go in /etc or %SystemRoot%/ProgramData

iandobbie, 2018-09-19:
So trying to code this now....

Windows
Data - c:\Data or read from config
Config - %SystemRoot%/ProgramData/Cockpit
Log - where?

Mac
Data -  ~/data or from config
Config - /etc/cockpit ?
Log - where?

Linux
Data - ~/data or from config
Config - /etc/cockpit
Log - where?

Can people please comment as this needs to be coded.

iandobbie, 2018-09-24:
Conclusions:

Config can define all directories

Defualts
Windows
data - C:\data
Log %localappdata%/cockpit
user config %localappdata%/cockpit

Linux:
data - cwd
Log ~/.cache - after check of xdg_cache_dir
user config ~/.config after check of xdg....

Mac:
I will check out

Config file path for Data

carandraug, 2019-01-22:
I have a patch to fix this for the case of the log files.  Please check my `logs-dir` branch.

```
git fetch david
git checkout david/logs-dir
```

Or to view he changes on the web interface https://github.com/carandraug/cockpit/compare/32f38733c382dd7421e74f9e9d7f216151d60006..febc74adc437cfd84503cf601a668869d17e039d

carandraug, 2019-01-22:
Got his reviewed in person by @mickp so pushed changes.

Because it might no longer be obvious where log files are going, Mick requested that at least until we're done with fixing this, also added a print statement during initialization. Also pushed that.

carandraug, 2019-01-23:
On the topic of MacOS, there is [macOS Library directory details](https://developer.apple.com/library/archive/documentation/FileManagement/Conceptual/FileSystemProgrammingGuide/MacOSXDirectories/MacOSXDirectories.html), a section of their File system programming guide. That link is for the archived version, I couldn't find it on the Apple latest documentation for developers.

iandobbie, 2019-04-02:
I think this is dealt with in Davids new config architecture.

closed 2019-04-02 by iandobbie