microscope-cockpit/cockpit #322: Repackaging broke cockpit on Windows.

mickp, 2018-07-12:
python setup.py develop

python cockpit
       can't find '__main__' module in 'cockpit'

Same result, regardless of where 'python cockpit' is called.

mickp, 2018-07-12:
OK. If I just do
 > cockpit
it starts, but it can't find FTGL.
Also, how do we fire up the .exe from the python debugger?

carandraug, 2018-07-12:
Once you run `python setup.py develop` there's a program named cockpit on your path. You don't need to specify `python cockpit` otherwise you are just passing the directory path as the first argument to python.

So:

```
$ python setup.py develop
$ cockpit
```

carandraug, 2018-07-12:
What I saw on Windows was that setuptools creates a `cockpit.pyw` next to `cockpit.exe` which I guess you could use. On Linux, only `cockpit` is created without file extension but that is a python script.

```
$ python -m pdb ~/.local/bin/cockpit
```

mickp, 2018-07-12:
Right. There's a 'cockpit-script.pyw' in the python Scripts folder. I guess this means I need to install for each virtual environment I need to test against, but that is also problematic.  

In python 2 virtual environments, setuptools complains that it can't find suitable distributions for wxPython (which is already installed) and futures (also installed). I know the dependencies are already met. Is there a way to force setuptools to skip the dependency checks?

carandraug, 2018-07-12:
I don't understand your issue but the answer to your question, how to install without checking for dependencies is

```
$ python setup.py develop --no-deps
```

mickp, 2018-07-12:
Without --no-deps:

`Processing dependencies for cockpit==2.9.0+dev
Searching for wxPython
Reading https://pypi.python.org/simple/wxPython/
Download error on https://pypi.python.org/simple/wxPython/: [Errno 1] _ssl.c:504: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version -- Some packages may not be found!
Couldn't find index page for 'wxPython' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: [Errno 1] _ssl.c:504: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version -- Some packages may not be found!
No local packages or working download links found for wxPython
error: Could not find suitable distribution for Requirement.parse('wxPython')
`

With --no-deps, it installs, but then when you run it:
`  File "C:\venv\cockpit\Scripts\cockpit-script.pyw", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\venv\cockpit\lib\site-packages\pkg_resources\__init__.py", line 3037, in <module>
    @_call_aside
  File "C:\venv\cockpit\lib\site-packages\pkg_resources\__init__.py", line 3021, in _call_aside
    f(*args, **kwargs)
  File "C:\venv\cockpit\lib\site-packages\pkg_resources\__init__.py", line 3050, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\venv\cockpit\lib\site-packages\pkg_resources\__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "C:\venv\cockpit\lib\site-packages\pkg_resources\__init__.py", line 969, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\venv\cockpit\lib\site-packages\pkg_resources\__init__.py", line 855, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'wxPython' distribution was not found and is required by cockpit
`

Yet wxPython is installed.

mickp, 2018-07-12:
OK - this looks like a setuptools/venv interaction thing, so nothing we can fix here. We just need to update the docs for the new installation method and usage - issue #323.

closed 2018-07-12 by mickp