microscope-cockpit/cockpit #342: Typos in experiment control fail with no error message.

iandobbie, 2018-07-18:
If you for instance have a z step size of 0.0.5 (as Maria just did) this generates a very opaque exception. We should sanity check this input before trying to run an experiment.

mickp, 2018-07-25:
Fixed for common experiment parameters in single site dialog window. Can use same validators elsewhere, but probably need to add a flag that allows blank entries for things like exposure times, where not every exposure time / light pair needs an entry.

mickp, 2018-08-28:
Added to more parameters in #382.
Some controls have duplicate names, because of the way they are created, so added colour to highlight which is the problem.
Validation is now applied to all common paramters for single-site experiments, namely:
  * rep number and duration
  * stack and slice heights
  * simultaneous and sequenced exposure times
  * bleach compensations in 2D and 3D SIM

I have not yet applied it to parameters that are specific to:
  * offset/gain correction file creation
    + number of exposures and collections
    + exposure multiplier
    + max intensity
    + cosmic ray threshold
  * response map creation
    + number of exposures and exposure times
    + max intensity
    + cosmic ray threshold
  * stuttered Z-stacks
    + intervals and reps per row
  * opto scripts
    + start and end for each row
  * rotator sweeps
    + settling time
    + V steps, start and max

mickp, 2018-08-29:
#384 applies float and int validators to more experiments. I think all the parameters are covered now, except those that expect a comma-separated list of floats.

mickp, 2018-08-29:
Comma-separated lists validated as ov b6e056f10047108676a666897400c6f3b1c1c575.

closed 2018-08-29 by mickp