microscope-cockpit/cockpit #345: datasaver only saves 16bit images

carandraug, 2018-07-18:
Saved files are always 16 bit unsigned integers. There's even a [TODO about it on the source]()

This is hardcoded in at least four places:

1. when computing number of bytes in a plane https://github.com/MicronOxford/cockpit/blob/b9d38c807b55a81979be8a0b6e96821c9466963e/cockpit/experiment/dataSaver.py#L145

2. when constructing the base header https://github.com/MicronOxford/cockpit/blob/b9d38c807b55a81979be8a0b6e96821c9466963e/cockpit/experiment/dataSaver.py#L218

3. creating the initial max and min values of the image https://github.com/MicronOxford/cockpit/blob/b9d38c807b55a81979be8a0b6e96821c9466963e/cockpit/experiment/dataSaver.py#L297

4. creating a padded version of the image https://github.com/MicronOxford/cockpit/blob/b9d38c807b55a81979be8a0b6e96821c9466963e/cockpit/experiment/dataSaver.py#L422.

In the case of the padded buffered image, it's not even clear why it's needed.

carandraug, 2018-07-18:
Fixed the one about selecting initial max and min values with commit cf3a466

I think the padded version of the image could be fixed by using the same data type as dataImage (actually, should be using `numpy.pad` or maybe not do any padding at all since the comment suggests that it shouldn't be required)

Not sure how to fix the other two issues, it needs to query the camera about the image bitdepth before experiment starts.

iandobbie, 2023-02-14:
This issue was discussed at the 2023-02-14 development meeting as some cameras can return 8 bit images. One suggestion is to enhance the dataImage with metadata. This would include bit depth or data type as well as other useful info like exposure time, excitation and emission wavelength, and any other data people think would be useful.