microscope-cockpit/cockpit #347: Making cockpit work under Linux

mickp, 2018-07-19:
Cockpit does not currently work under Linux. This issue provides a place to record the problems in getting it to run, and the fixes/workarounds.

mickp, 2018-07-19:
There are currently no wheels for wxPython under python 3, so pip/setuptools attempts to build it. This needs a *lot* of space in whatever temporary folder pip is using. I worked around it by remounting /tmp with:
`sudo mount -o remount,size=6G /tmp/`

Pointing TMPDIR elsewhere may also work, but I haven't tried it.
`$ export TMPDIR=$HOME/tmp`

mickp, 2018-07-19:
FTGL needs to be installed. I found a build in my package manager. 
`pacman -S ftgl`
Is there a way for setuptools to test for the presence of FTGL?

mickp, 2018-07-19:
It works! (under python 3.6.6, with wx 4.0.3 gtk3)
I can even run video mode with one or two cameras. 
But.
* It looks like crap.
* Main window is behind all the other windows, and can't be raised.
* Buttons in camera views window show a menu, but clicking on menu items just closes the menu - nothing else happens.
* Can't exit by closing the main window - clicking the close button does nothing.
* There are frequent segfaults from Pango, GTK and GLib.

mickp, 2018-07-20:
As of 5b9c3790e74a862f8176189168c70344b96f8f42, it's basically works on a VM running Manjaro. I can run single snaps, video mode, mosaics and experiments with microscope running dummy remotes.
* It still looks like crap.
* I see frequent critical errors from GTK and Pango on StdErr
* There are a bunch of deprecation warnings in the logging window, mostly concerning wx menus, I think.
* I see frequent TypeErrors from OpenGL, but they don't appear to be critical (at least on a VM).

carandraug, 2018-07-27:
> I imagine `apt-get install ftgl` would work under Debian-derived dists.

It does not.  You need to install `libftgl-dev`

---

Also, I fond out why cockpit segfaults during start in debian buster. The problem is not on cockpit or wxPython. Not sure where exactly the problem lies but seems to be related with wxWidgets, Wayland, and OpenGL. And Wayland is now the default in Debian and Fedora.

carandraug, 2018-07-28:
> Also, I fond out why cockpit segfaults during start in debian buster. The problem is not on cockpit or wxPython. Not sure where exactly the problem lies but seems to be related with wxWidgets, Wayland, and OpenGL. And Wayland is now the default in Debian and Fedora.

Seems that to work around the issue, one can set `GDK_BACKEND` like so:

    $ GDK_BACKEND=x11 cockpit 

So cockpit is now running on my laptop (Debian Buster)

This comes from [Debian bug #904753](https://bugs.debian.org/904753)

iandobbie, 2018-08-08:
Can we close this or are the weyland issues still there?

carandraug, 2018-08-16:
I was thinking of something along those lines but only if using the wxwidgets gdk, and then only of the user had not already specified the gdk backend to something.

carandraug, 2018-08-20:
> I was thinking of something along those lines but only if using the wxwidgets gdk, and then only of the user had not already specified the gdk backend to something.

The reason for this is to avoid messing with the environment unless we think is needed (only if platform is gtk) and still allow the user to set the backend himself (so only set GDK_BACKEND if not already set). I have the fix now.

The original bug though was for all linux related bugs, some of which are not yet fixed (namely, still looks like crap). Let us open an issue for each issue that comes. Closing this as fixed.

closed 2018-08-20 by carandraug