microscope-cockpit/cockpit #349: SIM diffraction angle is not vital metadata but not recorded.

iandobbie, 2018-07-20:
Changing the SIM diffraction angle changes the stripe width which is vital for recenstructions. We need to store this value somewhere. As this is only relevant in SIM experiments my initial idea is to store it in the dv file header after the collection order.

carandraug, 2018-07-20:
There is no place for that information on the metadata of the dv files. You can place it on the titles section of the base header. There's a total of 800 characters there (10 titles of 80 characters each). If you want to do this, you can do one of the two things:

1. add a new method `SIExperiment.generateTitles` which calls the parent method (`Experiment` has it) and then appends extra lines
2. whatever device handler knows about diffraction returns that text in its `getSavefileInfo` method.

iandobbie, 2018-08-14:
Problem is that the handler is actually an executor which really just knows about triggers not angles. I changed the structuredIllumination experiment to add it if there is an 'slm' device it reads the diffraction angle parameter and adds it when it adds collection order to the metadata #377

closed 2018-08-14 by iandobbie