microscope-cockpit/cockpit #363: Error checking on excitation wavelength.

iandobbie, 2018-07-26:
If you dont have a valid light-time pair, eg you select a laser that isnt on for an experiment, the code crashes in trying to set the excitation wavelength for the new metadata. We need some sanity checking rather than an opaque crash.

carandraug, 2018-07-26:
I have looked into this and it fails in this line in `cockpit.experiment.experiment`

    max_wavelength = max([l.wavelength for l,t in lightTimePairs])

because `lightTimePairs` is an empty list.

If we want to do sanity checking, then I'd guess it would go into `Experiment.sanityCheckEnvironment()` which experiment classes can overwrite.

Still, for that specific line, I would have:

    max_wavelength = max(0.0, *[l.wavelength for l,t in lightTimePairs])

so that the code does not fail if someone one day decides to have an experiment without a light source (or a light source that cockpit does not know about).  Does that sound ok?

carandraug, 2018-07-26:
mickp wrote:
> Fixed by 7f08e79

I thought the error was about checking if the light / time pairs are valid. If not, then we should fail the experiment, or inform the user about possibly bogus experiment settings. Or maybe I missunderstood what the issue is.