microscope-cockpit/cockpit #367: construction of abstract class `wx.TopLevelWindow`

carandraug, 2018-07-30:
Starting cockpit in my Linux system (Debian Buster, Gnome 3.28.2, display server Wayland but `GDK_BACKEND` set to `x11`, wxPython version 3.0.2, python 2) raises the following error:

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk3/wx/_core.py", line 16765, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/carandraug/src/cockpit/cockpit/cockpit/__init__.py", line 232, in doInitialLogin
    cockpit.util.user.login(wx.TopLevelWindow())
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk3/wx/_windows.py", line 403, in __init__
    def __init__(self): raise AttributeError, "No constructor defined"
AttributeError: No constructor defined
```

The reason is the line:

```
        cockpit.util.user.login(wx.TopLevelWindow())
```

where an instance of `TopLevelWindow` is constructed. However, this is an abstract based class. From [wxPython documentation](https://wxpython.org/Phoenix/docs/html/wx.TopLevelWindow.html):

> wx.TopLevelWindow is a common base class for wx.Dialog and wx.Frame.
>
> It is an abstract base class meaning that you never work with objects of this class directly, but all of its methods are also applicable for the two classes above.
>
> Note that the instances of wx.TopLevelWindow are managed by wxWidgets in the internal top level window list.

I guess that we need is the main window to use as parent for the user login window.

mickp, 2018-07-30:
That should probably wx.GetApp().GetTopLeveWindow().

carandraug, 2018-07-30:
I'm guessing we should be using [`wx.GetToplevelWindows`](https://wxpython.org/Phoenix/docs/html/wx.functions.html#wx.GetTopLevelWindows) and pick one from there. However, single this is only used to create a `wx.SingleChoiceDialog` which is a modal dialog so we can set parent to `None`

carandraug, 2018-07-30:
Further discussed this with Mick which agrees with reasons to set to `None`. It works in both my Linux system and Windows 7 (deepsim machine) so I pushed the fix. Closing as fixed.

closed 2018-07-30 by carandraug