microscope-cockpit/cockpit #369: Cannot change power on deepstar

juliomateoslangerak, 2018-07-31:
Hi I'm trying to merge all the changes you have brought to cockpit/microscope and get it to work in Montpellier. So I will post some of the issues where I'm blocked.

There are some issues with the lasers. My feeling is that there is some configuration that has been setup with the software and that is, somehow not override by microscope's initialization.

Do you have an idea of the problem? I'm running 

Server side everything seems to work: 
config is:
`device(deepstar.DeepstarLaser, '10.6.19.21', 9011, com='COM9', baud=9600, timeout=0.5)`
output is:
> 2018-07-31 09:21:37,413:DeepstarLaser:DEBUG:PID 2680: Debugging messages on.
2018-07-31 09:21:37,543:DeepstarLaser:INFO:PID 2680: Current laser state: [S1]
2018-07-31 09:21:37,563:DeepstarLaser:INFO:PID 2680: Device initialized; startin
g daemon.
2018-07-31 09:21:37,563:DeepstarLaser:INFO:PID 2680: Serving PYRO:DeepstarLaser@
10.6.19.21:9011

Client/cockpit side: Sorry for the mess. 
After Start up
>C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\re.py:212: FutureWarning: split() requires a non-empty pattern match.
  return _compile(pattern, flags).split(string, maxsplit)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\julio\pycharmprojects\cockpit\cockpit\handlers\lightPower.py", line 101, in _updater
    light.lastPower = queries[light].result()
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\concurrent\futures\_base.py", line 425, in result
    return self.__get_result()
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\concurrent\futures\_base.py", line 384, in __get_result
    raise self._exception
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\concurrent\futures\thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
IndexError: list index out of range

The GUI shows nothing wrong:
Deepstar 488nm Power / SET: 10.0mW / OUT: 0.0mW / OFF

Very often though I see question marks in stead of the power.

When I turn on (left click)
It turns on (green) and STOUT shows "Error in subscribed func lightSourceEnable in cockpit.gui.touchscreen.touchscreen."

I try to change the power to 50mW and:
>Exception in thread Thread-15:
Traceback (most recent call last):
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
IndexError: list index out of range

The output at the remote:
2018-07-31 09:31:16,022:DeepstarLaser:INFO:PID 2680: level=0
2018-07-31 09:31:16,022:DeepstarLaser:INFO:PID 2680: power=204
2018-07-31 09:31:16,022:DeepstarLaser:INFO:PID 2680: power level=PP0CC
2018-07-31 09:31:16,086:DeepstarLaser:INFO:PID 2680: Power response [PP0CC]
2018-07-31 09:34:24,956:DeepstarLaser:INFO:PID 2680: Turning laser ON.
2018-07-31 09:34:24,986:DeepstarLaser:INFO:PID 2680: Enable response: [UK]
2018-07-31 09:34:25,486:DeepstarLaser:INFO:PID 2680: L2 response: []
2018-07-31 09:34:25,496:DeepstarLaser:INFO:PID 2680: Enable-internal peak power response: [UK]
2018-07-31 09:34:25,996:DeepstarLaser:INFO:PID 2680: MF response []
2018-07-31 09:34:26,006:DeepstarLaser:INFO:PID 2680: Are we on? [UK]
2018-07-31 09:34:26,506:DeepstarLaser:ERROR:PID 2680: Failed to turn on. Current status: []

juliomateoslangerak, 2018-07-31:
It seems that laser power changes are working as long as the laser is OFF.
Still, changes are not reflected in cockpit's GUI.

mickp, 2018-07-31:
> It seems to be a timing issue. Adding a wait period to read back the
> answer from the laser's serial solves the problem.
> A wait of 0.1 sec is not enough.
> Maybe there is a real solution to this. So far I tested to rise the
> timeout of the serial connection with no success.
>
> Still, there are some some issues with some sequences of turning on and
> off the lasers, but I believe this is an error going through the Executor.

Sounds about right. The serial implementation on the Deepstar controllers
needs a long timeout - I've seen it stall for nearly 2 seconds before.

mickp, 2018-07-31:
It should be fine if you just set the serial timeout to 2s in the device
constructor in your config.
    e.g. device(DeepstarLaser, '127.0.0.1', 8000, com=3, baud=9600,
timeout=2)
Are you saying that doesn't work?

juliomateoslangerak, 2018-07-31:
Right. I tried a timeout of up to 10 sec!
device(deepstar.DeepstarLaser, '10.6.19.21', 9011, com='COM9', baud=9600, timeout=10)
The only thing that works is to insert a time.sleep(.2). Basically:

>def _write(self, command)
        command = command.ljust(14) + b'\r\n'
        response = self.connection.write(command)
        time.sleep(.2)
        return response

No problems what so ever with the OBIS implementation.
Could it be caused by the fact that the Deepstar is connected through a serial-USB adaptor...

I actually realized that everything was systematically working when I was using the debugger, which eventually gives that time...

iandobbie, 2018-08-08:
So is this issue fixed? Do we pull the sleep(.2) into the main repository?

Ian

iandobbie, 2018-08-09:
It turns on (green) and STOUT shows "Error in subscribed func lightSourceEnable in cockpit.gui.touchscreen.touchscreen."

This is an unimportant error. With microscope running the lasers this function isnt actually doing anything  as far as I can tell. I have fixed it on deepsim I'll check that the update gets to master.

juliomateoslangerak, 2018-08-10:
So what about the 0.2 sec wait time?
It seems to me like an odd situation in my setup... too odd to add it to master...?
I could try to connect the laser through the serial port directly (no usb adaptor)

mickp, 2018-08-16:
I don't really see what that wait time is doing. That line shouldn't be reached until the connection.write has either completed or timed out; either way, adding more wait time here should have no effect - response either contains a value, or is set to None after a timeout, and waiting 0.2s before returning it should not make any difference. All it does is increase latency, which is already high for the DeepStars because their comms is slow, and write can be called several times in any state change or query.

Julio - have you tried this with a version of microscope after MicronOxford/microscope@849316ec897132825068d0601fbe4714bc5be02a 
without your 0.2s delay?