microscope-cockpit/cockpit #371: Fresh Mac install fails.

iandobbie, 2018-08-13:
I tried on a fresh macos machine (10.12.5 - yes it is an old machine).
installed brew
brew install python3
git clone cockpit
python3 setup.py develop --user
crashes as it cant find libftgl.dylib

Fix with
brew install ftgl 

Then generates and error shappedbutton.py requires PIL
pip3 install Pillow fixes this issue.

carandraug, 2018-08-13:
These are two different issues:

1. Not specifying PIL as dependency. This is a duplicate of issue #319 which is not our bug, it's wxPython's. We have already fixed this in wxPython and will be part of their next release.

2. Not checking for ftgl native library during install. I'm unsure about this. setuptools and distools don't seem to have a method for this. We can add a check in `setup.py` but then we will have to be careful to only run it during install target or other targets that would actually require it. But because this check is not builtin in setuptools like the other checks of dependencies, it won't be run when starting cockpit. So maybe the fix is to handle the missing ftgl dependency more gracefully if it's missing at runtime (which I guess we should do since it's possible that ftgl is removed between install and run at a later date).

iandobbie, 2018-08-14:
We could easily wrap the code in a try:... except:

try: 
    if os.name in ('nt', 'ce'):
        _ftgl = ctypes.WinDLL('ftgl')
    elif sys.platform == 'darwin':
        _ftgl = ctypes.CDLL('libftgl.dylib')
    else:
        _ftgl = ctypes.CDLL('libftgl.so')
except:
    print ('Unable to open FTGL Library')

But I dont know what is best to do in  the except clause. Close? some dialogue box?

carandraug, 2018-08-14:
So the problem is that it's currently failing with `OSError` and an error message like "libftglas.so: cannot open shared object file: No such file or directory" which is not obvious being the missing library.

I think we should catch only the OSError, and then behave like any other missing dependency. If a python package is missing after installation, an `ImportError` is raised. Since this is not really an import we could raise a `RuntimeError` with a more suitable error message.

iandobbie, 2018-08-28:
Ok, should be fixed in #381

closed 2018-08-28 by iandobbie