microscope-cockpit/cockpit #389: Add timelapse counter to status lights.

mickp, 2018-09-04:


iandobbie, 2020-07-07:
As we have now deleted the status lights for a status bar I think this needs to be implemented in the status bar

carandraug, 2020-07-07:
There was the proposal of keeping the experiment dialog open until the experiment ends. This also addresses the issue of after starting the experiment not knowing which experiment was run (I remember the suggestion of triggering a screenshot and saving it on the desktop before starting the experiment so the user could inspect the image and know what it was running).

In that case, it would make sense to have the timelapse counter on the experiment dialog with a progress bar.

iandobbie, 2022-05-17:
I have an implementation that does this but it doesn't abort properly if the collection is cancelled and I don't understand why.  It is in 
https://github.com/iandobbie/cockpit/tree/timelaspe-timer
I will try to work out what the problem is a solve it properly.