microscope-cockpit/cockpit #391: Add ability to save snap image to file with metadata

mickp, 2018-09-04:
From either right click on image, or context menu on snap button that lists channels

iandobbie, 2021-03-19:
There is partial support, you can now save a snapped image but you dont get the meta data.

iandobbie, 2021-05-13:
This relates to issues #538,  #629 and #676 as they would all benefit from the additional meta data, howveer it is not trivial as we need to extend the image object to include this meta data. It would also help if we want to make colour images/mosaics.

iandobbie, 2022-07-14:
Saved snapped images now get pixel size and xyz position. We should probably also add the relevant extended header info as well as exposure time and active light sources probably in the free text fields.

closed 2023-10-10 by iandobbie