microscope-cockpit/cockpit #402: loading and resetting experiment settings fails

carandraug, 2018-09-27:
The 'Reset' button in the Experiments dialog fails with:

```
experimentConfigPanel.py", line 412, in onLoadExperiment
    panel = self.resetCallback()
  File "/home/carandraug/src/cockpit/cockpit/cockpit/gui/dialogs/experiment/singleSiteExperiment.py", line 136, in onReset
    self.sizer.Remove(self.panel)
TypeError: Sizer.Remove(): arguments did not match any overloaded call:
  overload 1: argument 1 has unexpected type 'ExperimentConfigPanel'
  overload 2: argument 1 has unexpected type 'ExperimentConfigPanel'
```

The same error happens when loading experiment settings from a file.

I'm running wxPython version 4.0.1.

carandraug, 2018-09-27:
Seems to be that 'wx.Sizer.Remove` does not accept a `Window` argument, we need to pass  its sizer. However, when I do that it segfaults:

```
corrupted size vs. prev_size
Aborted
```

I hope to get this fixed while I rework the rest of the experiment dialog.

iandobbie, 2022-05-17:
The reset now works and save appears to work but the load function fails with 

Traceback (most recent call last):
  File “/Users/ID/src/cockpit/cockpit/gui/dialogs/experiment/experimentConfigPanel.py”, line 403, in onLoadExperiment
    panel = self.resetCallback()
TypeError: onReset() missing 1 required positional argument: ‘event’

iandobbie, 2022-05-17:
Added a default event=None option which allows the load cal of the onReset function actually work and load previously saved experiments.

closed 2022-05-17 by iandobbie