microscope-cockpit/cockpit #412: Need to be able to store AO setup for logging and reuse.

iandobbie, 2018-11-06:
We need some functions and a file definition to allow dump and restore of AO settings.

iandobbie, 2020-07-07:
I believe that Nicks code now stores AO settings, not sure how easy it is to load them.

NickHallONBI, 2020-07-07:
At this point, I only store the aberrations applied once a sensorless correction has been performed. It's stored in a log file in the LocalAppData called 'sensorless_AO_logger.txt'.  Lines 876-885 of the cockpit/device/microscopeDeformableMirror.py deals with this. It stores the time stamp, the aberrations measured (and therefore corrected for) and the actuator positions applied. 

Currently, we don't store the actuator positions at any other time, but you can retrieve the last actuator positions applied by calling get_last_actuator_values() from the Pyro remote. 

The ROI coordinates, control matrix and system flat values are stored in the config.py file in LocalAppData.

iandobbie, 2021-03-19:
These data are stored, but I think at least the control matrix and possibly the system flat should be stored in the global rather than local config files. We have been bitten badly in the last week with not realising the control matrix is user specific and must be reloaded or regenerated for each user.