microscope-cockpit/cockpit #413: Movement step sizes need some work.

iandobbie, 2018-11-06:
There are a range of issues. 

1) XY and Z step sizes are linked and at a mad level. 

2) When the piezo stage is selected on deepsim you cannot move xy

3) The piezo step sizes are a bit mad too 1.25, 2.5,12.5. I think we need at least a 5.

iandobbie, 2018-11-07:
point 2 fixed in #415

NickHallONBI, 2018-11-14:
Tested on DeepSIM and it doesn't appear to break anything.

carandraug, 2018-11-14:
I have pushed 18baf6b which Nick just tested.  This fixes point 2 of this issue.

mickp, 2018-11-20:
Nope.  Commit 18baf6bc6439bbc1e4ea10c2fd74a5c completely breaks mosaic and macrostage: the stage is twice as large as it should be, moves move twice the distance that they should, and the mosaic is just plain weird.

Reverting.

[image]

carandraug, 2018-11-20:
Can someone now test my branch carandraug/cockpit@98f48250ad51b3577afa7448dd7282db939c1263 (branch refix-stagemover-413)

This ensures that we don't pick duplicated handlers when computing stage limits (adds all stages together) and movement sizes (also uses the non-used stages to compute offset).

carandraug, 2018-11-20:
@mickp tested this and approved. Reverted the reversion, rebased the fix, squashed, and pushed. This re-closes point 2 of this issue.

iandobbie, 2021-03-19:
This is a solved issue so closing

closed 2021-03-19 by iandobbie