microscope-cockpit/cockpit #416: Z stage getMovementTime

juliomateoslangerak, 2018-11-08:
Hi,
For the Z piezo, am I wrong or is the movement time specified as a constant of  5 msec independently of the delta?

During the z-stack experiment, the first image is taken while the z is still moving, probably because the first move is larger (I use the middle of the stack option).

mickp, 2018-11-12:
The SimplePiezo just sets it to 5ms, which is adequate for the piezos we
use on a number of systems. We could replace this with a rate, minimum and
maximum movement times, and default to the min. movement time if the config
has no rate defined. How long does it take your piezo to move through its
full range?

juliomateoslangerak, 2019-07-23:
This issue is fixed now

closed 2019-07-23 by juliomateoslangerak