microscope-cockpit/cockpit #421: Macro Stage Z: saved top bottom lines are only updated after a z movement

carandraug, 2018-11-21:
Changing the saved top and bottom positions does not update the Z macro stage view. Only after a z movement is a repaint triggered and the new saved positions are updated. To reproduce:

1. start cockpit
2. note that there's already saved top positions (text boxes on the Macro Stage Window) but these are not draw on the Z Macro Stage (no blue horizontal lines)
3.  make any Z movement
4. Z macro stage is redraw and saved top/bottom (horizontal lines) are drawn
5. click "Save Top". This updates the value in the text box but does change the blue line
6. make any z movement. Only now does the blue line showing saved position is updated.

carandraug, 2019-04-02:
There were two parts to this bug.

1. the saved top and bottom lines would are not draw at startip, only after the first movement. This was fixed as part of 3b27dc9c8bfc which removed the whole user login stuff.

2. saving a new top and bottom while running does not update the displayed top and bottom. For this, I think the saved/top panel needs to be a class on its own. The click of the buttons need to trigger an event that will redisplay the Z macro stage.

mickp, 2019-04-02:
Re. point 2 ...
On the base DeviceHandler, I added a watch mechanism so that UI elements can call some method to update themselves when handler attributes that they're watching change - this makes it easy to support multiple controls/displays for the same information. Is there a case for factoring this out as something that the base DeviceHandler and Interface classes inherit?

carandraug, 2020-04-13:
A quick and dirty fix is on 677594259566 (Ian's 'savetopbottom' branch).

iandobbie, 2020-04-14:
As David mentioned in #576 my dirty fix requires the button panel to track all instances of macrostage which is why I thought it was a nasty hack. Alternative suggestions welcome!

carandraug, 2020-05-14:
Thinking about this, I think that we should first move the concept of saved top/bottom to one of the interfaces module. At the moment, that information is on the `saveTopBottomPanel` which is a GUI component. This is not the place to store data, it's only the place to have a view on that data, especially when that data is meant to be accessed by other components (the experiment dialog also needs to get the saved top/bottom). Of the interfaces we have, the place to store this is the `stageMover`. Once we have this information in there, then we can just use an event to update their view, just like we do when we change the z step size or the z position.

iandobbie, 2020-05-15:
I agree. Having the information in the gui component is mad, and causing several problems.

carandraug, 2020-05-18:
I have a proposed fix on my `wip-421-saved-top-bottom-events` branch. It moves the saved top/bottom positions as attributes on the `StageMover`, refactors the saveTopBottomPanel on a proper Panel class, and then changes the logic to have the `StageMover` publishing an event for this, the same it does for when, for example, it changes the step size.

carandraug, 2020-06-01:
I have rebased and pushed the fixes suggested above caebbf9a6a20..0cdf9273ca81 since no one seems to have nothing against them. Closing as fixed.

closed 2020-06-01 by carandraug