microscope-cockpit/cockpit #423: mosaic autoscales images on import.

iandobbie, 2018-11-22:
The mosaic now autoscales images on import. It shouldn't do this, it should use the current displayed values from the camera view window. Now every images has a different scale and is impossible to compare.

mickp, 2018-11-23:
Not sure what's going on here. I've compared the current tile loading/saving code, and how it was over a year ago. The only changes are due to setuptools packaging and python 3 syntax changes.

The code does not seem to what you describe, and I'm not sure that what you describe is what we want: how would that work if you moved mosaics between systems where the cameras had different dynamic ranges?

The mosaic description file includes the black- and white-point used to display each tile. These are read back in and applied when the tile is reloaded, so the mosaic should be redisplayed exactly as it was when it was saved. This appears to be working, on both the saving and reloading sides.
  * if I edit the black and white points in the text file for a particular tile, the new scaling is applied to that tile correctly;
  * if I have a mosaic in which some tiles have different dynamic ranges, these are stored in the description file correctly.

If I save a mosaic and load it later, it looks exactly like it did when it was saved. I can rescale all the tiles using the camera range by right-clicking on 'rescale tiles'.

Do you have a mosaic file this doesn't work for?

iandobbie, 2018-11-26:
Sorry I wasn't clear in my original report. The moment the images are scaled is when they are taken and added to the mosaic, not imported from a file, sorry. The images used to come with the current scaling from the camera window. They now autoscale which means each image has a different scaling and is a pain.

mickp, 2018-11-26:
I can't reproduce that here: both the mosaic loop and transferCameraImage are fetching the scaling from the current camera with ```cockpit.gui.camera.window.getCameraScaling(camera)```, and the grayscale on mosaic tiles changes as I adjust the camera histogram brackets. If there's a problem fetching the camera window scaling, the mosaic stops and reports "Mosaic stopping - problem in getCameraScaling".

iandobbie, 2018-11-27:
This is very strange. I just tried it here at Diamond and the mosaic scaling worked as normal. Why it didnt work last week I have no idea! I will check during the day and if it doens't happen again I will close the issue.

mickp, 2018-11-27:
See what happens:
 * when there's more than one camera enabled;
 * when two cameras are enabled, then you disable one.
I don't see any reason why this should cause mosaic tile scaling to break ... but you never know.

iandobbie, 2018-11-27:
Further investigation shows that this pulls from the current image blackpoint/whitepoint, which are .1 and ,9 of min/max of the current image, NOT what the current viewer is displaying. So effectively this is autoscaling.

mickp, 2018-11-27:
Where does this happen? Which module/line?

mickp, 2018-11-27:
So it looks like it used to calculate this from the self.tiles[0][0] on demand, which could throw an indexing error if the first tile hadn't been populated yet. (And I don't think that would give different behaviour, because self.tiles[0][0] changes when a new image arrives.)
Instead, we'll save the absolute scaling values on the class at the time they're set using the histogram brackets.

iandobbie, 2018-11-27:
No the self.tiles[0][0] values are not recalculated on every newImage. The logic is a bit convoluted, but I just put in a pull request that adds a bunch of getCurrentScaling functions which reads these values rather than the latest image min/max. If first checks that self.tiles is populated so should not crash if there is no image. Then uses these functions in the mosaic get scaling calls. Maybe we should have a flag to return current or min/max for the image.

iandobbie, 2018-11-27:
I have a fix in #430

mickp, 2019-03-07:
OK - I've been looking at this again. The tiles stuff is there to handle images larger than 512x512, because this was the maximum texture size that the hardware at the time could handle. OpenGL has *required* support for textures of size 16384*16384 since at least GL4.1, released in 2011.
Do we expect to use a > 268MP detector any time soon?

mickp, 2019-04-07:
I think I finally fixed this with a rewrite of the imageCanvas stuff to bring the OpenGL code up to date and avoid repeated generation of textures for the same data, so it should be fixed when we merge mickp/glUpdates.

iandobbie, 2019-05-09:
I think this merge is complete and we just need to test. Is this true?

iandobbie, 2019-05-21:
This is fixed now with the mosaic updates.

closed 2019-05-21 by iandobbie