microscope-cockpit/cockpit #424: sr470 goes to sleep

iandobbie, 2018-11-22:
When the 561 line is activated then deactivated the shutter controller goes to sleep at Diamond. I have no idea why.

mickp, 2018-11-22:
Can't reproduce this at Micron: the SR470 works as planned, here. The shutter controller is brought out of sleep mode whenever the laser is enabled. 

I think there may be a fault with the shutter head at Diamond that prevents the controller from waking.

Power cycle the controller with the [Reset] key pressed. See if the behaviour persists.

Is the 'FAULT' light on the SR470 illuminated when you try to re-enable the laser? If so, there's a
problem with the shutter head, its power/control cable, or the 12V supply from the controller.

As a last resort, we could lose the
  self.send("ENAB 0")
from line 82 of sr470.py, but then the shutter might open unexpectedly when the DSP is initialised on
startup or the DSP machine reboots; this could be a safety problem if the laser had been left on.

iandobbie, 2018-11-22:
Managed to fix with this hack in devices/shutter.py

   def onLightSourceEnable(self, handler, enab):
        if enab and handler in self.lights:
            # One of our lights has been enabled.  Make sure that we respond to triggers.
            self.enableTrigger()
        elif not enab and not any([l.getIsEnabled() for l in self.lights]):
            # All of our lights are disabled.
        #    self.enableTrigger(False)
            pass

mickp, 2018-11-23:
That's equivalent to losing the "ENAB 0" command I mentioned above, and leaves the shutter responsive to spurious triggers from the DSP. Probably better to fix this in the sr470 device file, because then it only affects the shutter type that has the problem.

mickp, 2019-04-07:
Did we ever get to the bottom of this? I think we've only ever seen this behaviour at Diamond.

iandobbie, 2019-04-08:
Nope, not properly investigated. Fixed by a hack of removing the disable call.

iandobbie, 2020-04-07:
Diamond have ordered a new shutter controller on the assumption that this is a controller issue, especially as it was becoming unresponsive increasingly often. Will update once we can access the lab and test.

carandraug, 2020-05-21:
To be specific, the patch mentioned above by @iandobbie and was found to fix the issue, has not been pushed.

mickp, 2020-05-21:
Note the safety-related issue this patch may introduce (see comment dated 2018-11-23).

iandobbie, 2020-07-07:
We believe that this was due to hardware issues on the Diamond system. We have replaced the controller. I will ask Maria if this appears to be solved or not.

iandobbie, 2021-05-13:
I understand this is now solved and I am just going to close the issue

closed 2021-05-13 by iandobbie