microscope-cockpit/cockpit #425: SI experiment is broken at diamond....

iandobbie, 2018-11-22:
I'm pretty sure this is a python 2- versus python 3 issue. 

How can experiments be broken? we need to test this on deepsim in oxford before we check in any changes to dsp or experiment code.   

Traceback (most recent call last):
  File "c:\b24\cockpit\cockpit\gui\dialogs\experiment\singleSiteExperiment.py", line 130, in onStart
    if self.panel.runExperiment():
  File "c:\b24\cockpit\cockpit\gui\dialogs\experiment\experimentConfigPanel.py", line 520, in runExperiment
    return self.runner.run()
  File "c:\b24\cockpit\cockpit\experiment\experiment.py", line 218, in run
    self.createValidActionTable()
  File "c:\b24\cockpit\cockpit\experiment\experiment.py", line 276, in createValidActionTable
    self.table = self.generateActions()
  File "c:\b24\cockpit\cockpit\experiment\structuredIllumination.py", line 310, in generateActions
    curTime = self.expose(curTime, cameras, lightTimePairs, angle, phase, table)
  File "c:\b24\cockpit\cockpit\experiment\structuredIllumination.py", line 396, in expose
    curTime += delay
TypeError: unsupported operand type(s) for +=: 'Decimal' and 'float'

mickp, 2018-11-22:
For now, replace delay with Decimal(delay).

mickp, 2018-11-22:
This is because getMovementTime on the SLM resolves to a float. We need to decide how best to handle this: do we insist that all movement times are Decimals on the device, or do we just ensure that any floats are converted to Decimals in the experiments?

The point of using Decimal is to prevent the accumulation of floating point errors as we add up loads of small values --- from the point of view of the arithmetic, it doesn't matter where the small value is converted to a Decimal, as long as it happens before the addition step.

If we do the conversion in the experiment, we have to do it many times, but we ensure that we don't hit this problem again. (And there's a good chance that the python interpreter will optimise things so we don't really end up doing the conversion many times.)

If we do the conversion in the device class, we only do it once, but we'll hit this problem again with any device that forgets to do it.

mickp, 2018-11-22:
Sorry: it looks like I introduced it when I made changes for Nick's AO stuff a month or so ago.  It's not a python 2 to 3 thing. 

I think we should convert the values to Decimal when they're parsed from the config files: that way, '0.1' in the config file becomes Decimal(0.1), rather than Decimal(0.1+/-Epsilon). This is a pain, though, because it's easy to miss, and we have no automated test for it.

PR #427 (2018-11-22): Movement times must be Decimal.
Should fix #425.
closed 2018-11-22 by mickp