microscope-cockpit/cockpit #429: Automated storage of mosaic images

iandobbie, 2018-11-22:
On CryoSIM it would be great to be able to automatically store the mosaic images along with XY coords so we can reconstruct the mosaic to match with the xray images

mickp, 2018-11-23:
I don't understand what you mean. If you save a mosaic, the images are saved in a single MRC file, and the XY (and Z) locations for each image are stored in a metadata file. What more is needed?

iandobbie, 2018-11-26:
Maria asked for the ability to have the files autosaved as they were
collected, so the images saved as individual mrc files, probably with
the xyz fields populated in the mrc header and then stored in a 
separate mosaic sub directory. 

Just so that the mosaic data is not just junked if you forget to save it
explicitly.

mickp, 2018-11-26:
OK. How should we handle the following?
* Filenames - where should it go, and how to we identify which files belong to which mosaic?
* Tile deletion - do we delete the data?
* Multiple images at the same / overlapping locations.
* Cleanup - do we ask about deleting the dumped mosaic data on exit?

iandobbie, 2023-10-18:
I think this is not easily achievable. My recent work on extending the metadata in snap and mosaic saves has shown that the current dv save format cannot accommodate an unknown number of images. It would involve either a) leaving a large gap at the start of the file to accept the data retrospectively, or to write the file then build the correct header and rewrite the file after the mosaic has finished, and this doesn't account for continuing mosaics. 

I am going to close this as not realistically achievable. We can revisit this if we move onto a new file format, eg the work I have started on OME-zarr.

closed 2023-10-18 by iandobbie