microscope-cockpit/cockpit #432: Macro Stage XY: "Recenter" recenters both Z-stage and XY stage

NickHallONBI, 2018-11-28:
Just discovered that the recenter button in the Macro Stage XY will recenter Z-stage (as it should) but it will also recenter the XY stage. This problem occurs in my master branch and I believe this is related to merging David's latest stage fixes into that.

carandraug, 2018-11-28:
Proposed fix at carandraug/cockpit@29fe70d334057f11f997c48dc3f4777bc147d2be

To test:

    git checkout -b david-recenter-issue-432 master
    git pull https://github.com/carandraug/cockpit.git recenter-issue-432

To merge:

    git checkout master
    git merge --ff-only david-recenter-issue-432
    git push origin master

NickHallONBI, 2018-11-28:
Yep, this seems to fix it. Consider this issue closed.

closed 2018-11-28 by NickHallONBI