microscope-cockpit/cockpit #434: Image fully saturated error

juliomateoslangerak, 2019-01-09:
When the image is fully saturated (or for that matter all pixels have the same value) there is a 0 division error in line 317 of gui/imageViewer/viewCanvas.py

I could solve the bug by summing 1 to datarange in line 314.

`        dataRange = temp.max() - temp.min() + 1
`
Is that a sensible solution?

mickp, 2019-01-10:
Maybe? I think I'd prefer to do the +1 in the next line:
```python
numBins = min(dataRange+1, MAX_BINS)
```
It has exactly the same effect, but I think it shows the reasoning more clearly: dataRange is still what the name implies, and numBins is set accordingly --- single-valued data has zero range but one bin for the single value; for ints, a dataRange of 1 means 2 values, so 2 bins.

iandobbie, 2019-05-09:
Seems to me this file is largely rewritten, and I assume fixed? Can we close this issue?

closed 2019-07-11 by juliomateoslangerak