microscope-cockpit/cockpit #435: setHistogram menu bug

juliomateoslangerak, 2019-01-09:
gui/imageViewer/viewCanvas.py bugs when I try to change the histogram parameters using the right click contextual menu.
The issue being in lines 585 and 589 where a map object is being subscripted.
Replacing the mapping of the values in line 582 by a comprehension does the trick

`        values = [float(v) for v in values]
`

juliomateoslangerak, 2019-01-09:
I actually added a 
`        divisor = max(divisor, 1.0)
`
to avoid a zero division error too

mickp, 2019-01-10:
Yep:
 * List comprehension makes sense, since values[] is short.
 * A lower bound on divisor is sensible.

iandobbie, 2019-02-21:
These two issues are fixed in my branch histMenu, can someone check I haven't done anything mad.

carandraug, 2019-02-21:
For future reference, what happens when changing histogram parameters is the following error is thrown:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 559, in <lambda>
    self.Bind(wx.EVT_MENU,  lambda event, action = action: action(), id= id)
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 588, in onSetHistogram
    self.blackPoint = (values[0] - self.imageMin) / divisor
TypeError: 'map' object is not subscriptable
```

mickp, 2019-02-21:
It's a python 2-3 thing, I think. Iterating over the result of a map() call
will work in python 2, but it needs converting into an iterable in python 3.

carandraug, 2019-02-21:
Now pushed to master.

closed 2019-02-21 by carandraug