microscope-cockpit/cockpit #445: cockpit.devices.ni_cRIOFPGA fails to import due to missing cockpit.config variables

carandraug, 2019-01-22:
The module `cockpit.devices.ni_cRIOFPGA` fails to import because `cockpit.config` does not have the variables `LIGHTS`, `CAMERAS`, and `AOUTS`:

```
$ python3 -c 'import cockpit.devices.ni_cRIOFPGA'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "cockpit/devices/ni_cRIOFPGA.py", line 79, in <module>
    from cockpit.config import config, LIGHTS, CAMERAS, AOUTS
ImportError: cannot import name 'LIGHTS' from 'cockpit.config' (cockpit/config/__init__.py)
```

The removal of those variables hapened in 91c71979feb648e4c5d09d8bdbdd3329 which is more than one year ago. While that commit removed those variables, the commit message says that they were obsolete so maybe it didn't even work properly back then, despite the variables existence.

carandraug, 2019-01-22:
Bringing @juliomateoslangerak to the conversation since he's the one who actually has this board.

carandraug, 2019-03-08:
@juliomateoslangerak are you saying that you are going to do that polishing or should we pull it now? If the later, from where?

juliomateoslangerak, 2019-03-14:
I'm doing the polishing now. I will submit a pull request in a few days.

carandraug, 2019-03-14:
>  I will submit a pull request in a few days.

Just mention here which branch to pull the changes from.

juliomateoslangerak, 2019-04-12:
Hi,
Done. You may pull from my master branch. 
You should get the ni_cRIOFPGA, obviously, and I also made some slight modifications in executorDevices.ExecutorDevice.
- nr of analogues and digitals is a configuration property
- renamed DSP for Executor in a few places
- added a _adaptActions method to encapsulate all the executor-specific modifications of the actions table
- an in-situ list creation of a zip operation that was not working for me.

carandraug, 2019-06-05:
There were several other changes on your master branch which I didn't want to merge, or at least not together with the rewrite of the ni_cRIOFPGA module. So I just made a new commit with your whole ni_cRIOFPGA module with you as author.

closed 2019-06-05 by carandraug