microscope-cockpit/cockpit #446: can't define configurator options in the conf file

carandraug, 2019-01-25:
There is a `Configurator` device which is used among things to define the directory for the logs and userConfig files. However,  changing or adding adding options to this device requires actually modifying the source in `cockpit/devices/configurator.py`

There is a single `Configurator` device which is initialized at the start of cockpit. This is done as part of `depot.initialize` before reading the conf files that define other devices. Trying to define another `Configurator` device on the conf file, fails with:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/__init__.py", line 131, in OnInit
    for i, device in enumerate(depot.initialize(cockpit_config)):
  File "/home/carandraug/src/cockpit/cockpit/depot.py", line 347, in initialize
    for device in deviceDepot.initialize(config):
  File "/home/carandraug/src/cockpit/cockpit/depot.py", line 206, in initialize
    self.initDevice(d)
  File "/home/carandraug/src/cockpit/cockpit/depot.py", line 287, in initDevice
    self.addHandler(handler, device)
  File "/home/carandraug/src/cockpit/cockpit/depot.py", line 267, in addHandler
    (handler.name, str(device), str(otherDevice)))
RuntimeError: Multiple handlers with the same name [configuration] from devices [<cockpit.devices.configurator.Configurator object at 0x7f320a2e1710>] and [<cockpit.devices.configurator.Configurator object at 0x7f321e1b1128>]
```

The error comes from the fact that each `Configurator` device has a `ConfiguratorHandler` always named `configuration`. Depot will fail to initialize the second `Configurator` because there's already another handler named `configuration`.

I spoke with Mick about this. Seems like the whole Configurator class is mainly historically and we should be able to do away with it.

carandraug, 2019-01-28:
Currently, the configurator can be used for the following:

* `configDirectory`, `dataDirectory`, `logDirectory`: define the directory for log, config, and data files
* `slideTouchdownAltitude`: (z) position used for the "Touch Down" button in the macro stage window
* `slidealtitude`, `dishaltitude`: the z positions to draw the two peaks on the Z macro stage (one is supposed to represent a slide altitude and the other the dish altitude)
* `loadPosition`, `unloadPositiontouchdown`: load and unload position in the touchscreen (controls where in z the stage moves after pressing the "Load/Unload" button)

Currently, the only to change these files is by modifying the source in the `Configurator` device.

We could remove this class and have this be defined on a conf file that cockpit reads at startup.

carandraug, 2019-02-13:
I have branch `cockpit-configs` on my clone that addresses this issue. It comes with a bunch of previously discussed changes to how config is loaded. It is [this commit](https://github.com/carandraug/cockpit/commit/3cb89e112d736cc093b5de444839e985646817a9)

* adds cockpit config and depot config
* add command line options to specify both cockpit config and depot config
* stops depot config from being all conf files inside the config subpackage
* uses default config paths mentioned on issue #320 
* removes the config singleton (well, it is now part of the App itself)
* removes configurator device and handler
* adds a new dependency on the package `configparser` (python2 package that provides a configparser compatible with python3)

I have just tested it on deepsim and seems to work fine. Without anything special, it will do the following:

1. reads `%ProgramData%\cockpit\cockpit.conf` for cockpit configuration.  This may specify what files to use for depot
2. reads `%ProgramData%\cockpit\depot.conf`for depot configuration (or whatever files are mentioned on the cockpit configuration

I'm not adding the option to read all conf files in a directory by design to avoid issues where the order of device initialization will cause issues.

mickp, 2019-02-13:
Device order in the config files should not be an issue - users may not
know what order devices need to be initialised, so we can't expect them to
order them correctly in the config. There used to be a priority system, but
I think I completely replaced this with a dependency list: depot figures
out dependencies from config parameters (like 'analogsource' or
'digitalsource'), and pushes devices further back if their dependencies are
not yet initialised.

carandraug, 2019-02-14:
> Device order in the config files should not be an issue - users may not know what order devices need to be initialised, so we can't expect them to order them correctly in the config.

So far, devices that need other devices are the ones that mention them by name on the configuration. So the user that writes the configuration should know that they are dependencies.  For example, if a user writes `analogSource: dsp` in a device configuration he should be aware that a device named `dsp` needs to already exist. With this change, the user will also need to know that the device `dsp` must be defined before.

>  There used to be a priority system, but I think I completely replaced this with a dependency list: depot figures out dependencies from config parameters (like 'analogsource' or 'digitalsource'), and pushes devices further back if their dependencies are not yet initialised.

This only works as long as the only parameters that introduce dependencies are the keys `triggersource` and `analogsource`. But we now support other devices, maybe not be in cockpit, and we don't know what other keys they will have that may introduce dependencies.

---

We previously had no method to specify files with device configurations, it was all the conf files in the config subpackage. Because of that, there was no control over the order the files were read, and because of that, we needed to figure out the order within cockpit. At least that's how I rationalized the current situation.

But if now we can define the order of files and devices, should we still keep the option to include a directory and not have a order well defined in the config? It's just a few more lines of code to add that.

mickp, 2019-02-19:
The only things that have dependencies right now are things that have a trigger source or an analogue source. Any other kind of dependency would either:
 * be handled in microscope, by composing virtual devices from simple, real devices;
 * or involve us introducing a new dependency keyword.

Either way, this is something we can handle straightforwardly in the code, and I don't yet see a good reason to push a requirement onto the user that they ensure config entries appear in a certain order. Yes, they need to ensure that 'dsp' exists if something uses 'dsp', but they shouldn't have to consider which comes first in the config file.

carandraug, 2019-02-19:
Me and @mickp  spoke in person about this.  Follows a summary of what we spoke about, but the conclusion is that I will push the changes since they don't actually break the ability to define devices out of order.

---

This change does not require the user to specify the devices in order. The thing it does, is remove the ability to specify a directory and read all conf files there. The code that delays initializing devices until after its dependencies are initialized is still there. For me this two things go hand in hand, but we agreed that a user may want cockpit to figure out the order automatically.

The problem of future devices, not part of cockpit, with a dependency other than `triggersource` and `analogsource`, should not happen because the plan is to move all such issues from cockpit devices into microscope.  With time, cockpit devices should disappear or become very thin shells.

carandraug, 2019-02-28:
The changes to cockpit config have now been merged after trying on DeepSIM, and approval from mick, ian, and nick.

In the specific case of DeepSIM, the default system-wide depot configuration is `C:\ProgramData\cockpit\depot.conf` which I already placed with the help of Nick.

If you want to have any device also loaded by default (for example, Nick always wants his alpao device always loaded by default), then add their configuration in `C:\Users\USERNAME\AppData\Local\cockpit\depot.conf`.

If you always want to have a different set of devices loaded, then write their configuration in a file wherever, and then add to `C:\Users\USERNAME\AppData\Local\cockpit\cockpit.conf`:

    [global]
    depot-files: path-to-that-file
        path-to-other-file-maybe

If you want to run cockpit only a few times with a different set of devices, put them in a config file wherever and specify it with a command line option, like so:

    cockpit --depot-file path-to-that-file

I'm closing this now. I will push the documentation in another config, as part of issue #454

closed 2019-02-28 by carandraug