microscope-cockpit/cockpit #448: Need to account for binning and ROI in mosaic positions

mickp, 2019-01-25:
I recently implemented support for binning and ROI for the Andor CCD cameras in MicronOxford/microscope. Image display required a minor tweak to the imageViewer to track texture size and check against new image size. The mosaic will be a bit more involved.

When images are added to the mosaic, they are *centred* on the current image position. This means that a camera ROI that is not symmetrically cropped (so the image centre corresponds to the detector centre) will be placed at the wrong position on the mosaic. We'll need to check ROI and adjust the image placement, and I suspect we'll need to account for any image transforms on the camera side, too.

I will implement ROI and binning on the MicronOxford/microscope test camera so we can test this easily without the need for hardware.

iandobbie, 2019-05-09:
For this to work we need to move pixel size from the objective to the camera, otherwise what happens when even identical cameras have different binning?

iandobbie, 2020-04-07:
The mosaic basically works, Some strangeness but binning breaks the mosaic image positioning. I suggest we open a new ticket as the problem is really the pixel size goes with the camera image not the objective. 

Also relates to saving snap images #391