microscope-cockpit/cockpit #449: run script after image acquisition

carandraug, 2019-01-29:
There's a request to be able to have cockpit automatically perform some image processing after it is done with the acquisition. This happens in softworx which can be configured to automatically deconvolve the images after it is done acquiring.

This was requested to support an experiment type meant for quality control of microscopes. A facility manager would run a specific experiment that would run some processing and present the results at the end.

This could also be used to have the system automatically import the images to omero after acquisition.

This should be possible after the reword of the experiment window.

iandobbie, 2019-01-30:
Overall the idea is to be able to automate acquisition and analysis of bead images. Suggested plan is 

1. Find isolated beads
2. Acquire multi colour bead stack
3. Store data, possibly upload to OMERO
4. Run analysis, maybe command line PSFj
5. Capture output and collate for tracking

step 4 and 5 might be triggered to run in OMERO

iandobbie, 2022-05-18:
The final part of the cleanup after experiment thread is:

        events.publish(events.CLEANUP_AFTER_EXPERIMENT)
        if self.initialAltitude is not None:
            # Restore our initial altitude.
            cockpit.interfaces.stageMover.goToZ(self.initialAltitude, shouldBlock = True)
        events.publish(events.EXPERIMENT_COMPLETE)

So we could easily run a specified function by adding to the trigger on either CLEAN_UP_AFTER_EXEPRIMENT or EXPERIMENT_COMPLETE. Maybe we just need to code an example in the docs to show how this could be done.