microscope-cockpit/cockpit #451: System should calculate optimal Z spacing for Stacks.

iandobbie, 2019-01-31:
If we add NA to the objective definition we can calculate the expected depth of focus and recommend a Z slice depth. We could also warn if the camera pixel size is either over or under sampling by for than some factor, say 2

carandraug, 2019-01-31:
And this would also be a good default in the experiment panel.

iandobbie, 2019-02-01:
Added NA to the objective data in #455. Unfortunately we also need the refractive index of the immersion media to work out a sensible step size. The formula is 
wavelength*n/(NA^2) 

where n is the refractive index of immersion media. 

Adding this as well to the objective spec is getting silly. We need a better way to specify all the data required for each objective. Can we read in a python dictionary from the config file?

iandobbie, 2019-02-01:
Once we have the info a nice calc for the step size might be

int(wavelength*n/(NA^2)/100)*50

which gives resolution/2 in 50 nm steps rounded down.

iandobbie, 2023-08-18:
The new(ish) objective spec makes adding NA to the objective spec easy. I will implement this and then think about how best to provide calculated Z resolutions for stacks.

iandobbie, 2023-08-23:
Started to add this in my branch add-NA. This adds  NA and immersionRI to the objective spec and functions to calculate XYresolution and Zresolution to the objective handler if given a wavelength. 

We could add this info to the camera views and have a button to set ideal Z stack spacing to the z-stack experiment.

carandraug, 2023-08-23:
Why have the refraction index of the immersion oil a property of the objective when it is not?  Why not have a list of immersion oils somewhere and select one on the experiment window? Or maybe a text box to enter the RI manually in the experiment window?

iandobbie, 2023-08-23:
The immersion media is a property of the objective and is needed to calculate the Z resolution. Small changes make little difference so I am not thinking about the changes to reduce spherical aberration on a DV or OMX system, this is the difference between an air lens at 1.0, water at 1.32, silicon at 1.4 and oil at 1.52. There are a very few multi immersion lenses with a correction collar but they could have multiple depot entries for different media.