microscope-cockpit/cockpit #456: objective configuration is too complicated

carandraug, 2019-02-04:
The configuration for an objective is a single line with multiple values. Like so:

```
[objectives]
type: cockpit.devices.objective.ObjectiveDevice
40x: 0.2 ; 0 ; (0,0,0) ; (-100,50,0) ; (1,1,1)
60xWater: 0.1 ; 0 ; (0,0,0) ; (0,0,0) ; (1,1,1)
```

And now we want to add further values to each objective (see pull  #455) which will make configuration of each messier. We could have the configuration value be a dict, like so:

```
[objectives]
type: cockpit.devices.objective.ObjectiveDevice
40x: {'pixel_size' : 0.2, 'offsets' : (-100, 50, 0)}
60xWater: {'pixel_size' : 0.1}
```

or maybe even use nested sections (but that would require changing from Python's `ConfigParser` to something else like `ConfigObj`).

Alternatively, can't we instead have multiple objective devices, each with their own objective handler? Like so:

```
[40x]
type: cockpit.devices.objective.ObjectiveDevice
pixel_size: 0.2
offsets: (-100, 50, 0)

[60xWater]
type: cockpit.devices.objective.ObjectiveDevice
pixel_size: 0.1
```

carandraug, 2019-02-04:
The implications of having one device per objective is that we will then have a single objective handler per objective. I looked at the objective handler and it looks like it should have been `cockpit.interfaces.objectives` since it abstracts away all objectives in a single place, kinda like the imager and the stage interfaces.

mickp, 2019-02-04:
Yeah - it makes more sense as an interface than a device. Then the device
can just parse the config and add itself to the interface. The objective
config definitely needs cleaning up, as it's completely incomprehensible to
anyone that can't read the code: the one section per objective approach
looks fine to me.

mickp, 2019-04-26:
A new `cockpit.interfaces.objectives` class is definitely the way to go. Currently, 11 modules use `depot.getHandlersOfType(depot.OBJECTIVE)[0]` in 18 places, and one of these is a base class that stores the a local reference to the handler on itself.

dstoychev, 2020-08-14:
I agree that objectives should have their own sections in the configuration, but the section name should not be used to store important information such as magnification and immersion, else you end up having to parse it. Here is a complete example config:

``
[MyLongWorkingDistanceObjective]
type: cockpit.devices.objective.ObjectiveDevice
magnification: 60
immersion: Special
pixel_size: 0.2
lens_id: 0
transform: (0, 0, 0)
offset: (0, 0, 0)
colour: (0, 0, 0)
``

carandraug, 2020-08-14:
> [...] the section name should not be used to store important information such as magnification and immersion, else you end up having to parse it.

That is not being proposed here. The magnification and immersion are the objective/device name because that's how user refers to them. The rest of cockpit does not make use of the magnification or the immersion information, therefore it does not parse it, it's really only the name. If (or when) cockpit ever makes use that information, I agree, those should be configuration values. But even then, chances are that objectives will still be named something like `40xWater` or similar.

iandobbie, 2020-08-14:
I think David's comment is correct, the information in the name is for user reference. The the critical factor becomes what properties should the objective have, and what is compulsory and what is optional?

I suggest:
compulsory
magnification:
pixel size (see note below):
NA: 
lensID:
immersion:

optional
offset: default (0,0,0)
colour: default (0,0,0)
transform: default (0,0,0)

Pixel size is an issue at the moment. It is currently set by the objective but may be different on different cameras so should be a property of the camera, but is related to the objective. Also not clear how to store this in output files as thet typically have a single pixel size for every image in a multi-image file.

juliomateoslangerak, 2020-08-19:
I agree with Ian's suggested compulsory/optional config variables. Maybe I would consider the serial number on top of lensID.

On the pixel size issue, maybe there could be a camera pixel_size correction factor defaulting to 1. Indeed in most cases a multichannel image would have equivalent pixel sizes but these are things that might change. 

There is an additional case when we consider the use of additional 'slide-in' optical components. 'Optivar' lenses will modify pixel size.

iandobbie, 2020-08-19:
Objective serial number is a good idea. Still not sure how to do the pixel size, in part because storing the data becomes hard as file meta data typically has one pixel size. Camera correction factor could work, but if we are doing this why not just set the pixel size per camera. We could have a dictionary with {objective name: pixel size} per camera. Deciding how to save this data is another problem.

carandraug, 2020-08-21:
I started working on this with the initial proposal discussed with Mick, i.e., have each physical objective be its own section/block on `depot.conf` (instead of each physical objective being one of the values in the objectives device), then each objective device has a corresponding objective handler, and an objective interface handles the current interface so the gui components interact only via the interface.

However, I found that this is problematic for a multi view system (see #554). In theory, cockpit could support this. The `ObjectiveDevice` is currently a collection of physical objectives, so you can have multiple `ObjectiveDevice` each with their set own of objectives. The main panel also places the objective dropdown menu in different places if you have more than one. But that's it, nothing else in cockpit supports it and ignore the rest:

* `depot.getHandlersOfType(depot.OBJECTIVE)[0]` is used to get the current objective so only the current objective in the first objective device is used all throughout cockpit
* both `piDIO` and `ni6036e` only pick the `ObjectiveDevice` named `objective` (which also is a problem if the device is named something else)
* `MicroscopeCamera` will update the transform if any objective change
* the data saved from an experiment also only checks one objective

So I have an almost finished patch (see my branch `wip456-objective-interface`) but it removes the little start of support cockpit has for a multi view. I'm unsure if it's worth keeping that support anyway since it's obviously not being used, and I'm not sure the rest of cockpit infrastructure can deal with multi-view without a lot of work. Opinions?

By the way, why is transform a property of the objective?

Also, I previously said that cockpit does not parse the objective name to use its magnification and immersion. I've found that it is not true, it parses the magnification to display them sorted by magnification (see `cockpit.handlers.objective.ObjectiveHandler.sortedObjectives`).

carandraug, 2020-08-25:
> [...] removes the little start of support cockpit has for a multi view. I'm unsure if it's worth keeping that support anyway since it's obviously not being used, and I'm not sure the rest of cockpit infrastructure can deal with multi-view without a lot of work. Opinions?

We spoke about this on Monday's meeting. @iandobbie agreed on moving forward with the initial proposal and removing the little and incomplete support for the case of multi set of objectives, each of them to different cameras.

> By the way, why is transform a property of the objective?

This is because each different objectives are actually used for different light paths and changing an objective might also mean going through some other path with extra mirrors that may flip the image.

carandraug, 2020-08-27:
As discussed, I pushed the changes to simplify the objective configuration. I did not change the objective fields, neither changed which ones are required and which ones are optional. But with the work from 7e08699 merged, that should be trivial.

iandobbie, 2021-03-19:
This is all merged and working so closing the issue,

closed 2021-03-19 by iandobbie