microscope-cockpit/cockpit #462: cockpit status after changes to config

carandraug, 2019-03-08:
The `cockpit.scripts` module is actually a script. With the latest changes to `cockpit.config` it no longer works.  Because the actual devices used are dependent on command line options, there may be a disconnect between its output and the actual devices in a running cockpit session.

I think the following two things can be done:

1. add a way to run it from inside cockpit itself. This is the easiest way to ensure that we use the same configuration.
2. add a `cockpit-status` script that takes the same command line options as `cockpit

carandraug, 2019-03-20:
I have pushed the minimal changes to make the script it work again as discussed on the lab meeting.

Because the config is dependent on the environment and command line options, the output of `cockpit/status.py` only makes sense if under the same conditions. It takes exactly the same command line options as cockpit itself.

iandobbie, 2019-05-09:
it is intended as a quick and dirty hardware diagnostic. Seems to work fine on DeepSIM. Can we close the issue?

closed 2019-05-10 by carandraug