microscope-cockpit/cockpit #467: nicer window display for exceptions

carandraug, 2019-04-04:
If there is an exception initialising cockpit, the traceback is show in a StaticText widget, with"normal" font family, and text wrapping.  The problem of static text is that we can't select the text to copy and paste somewhere else (future user googling for users, or grepping the code). The non monospaced text and text wrapping makes the traceback more difficult to follow.  Currently, it looks like this:

[image]

As part of another work, I wrote a simple function to display an exception traceback. It looks like this:

[image]

There is also a version where the traceback is in a collapsible pane (collapsing/uncollapsing can be done via keyboard with Tab + Space):

[image]
[image]

This is a function meant to be used wherever in the cockpit gui when an exception occurs. At the moment, the only place is during initialisation. On the work I have done for experiment panel, I also use when an experiment fails during initialisation.

I have an `exception-box` on my clone.  Or see the commit at https://github.com/carandraug/cockpit/commit/cbf384528ab1f32caedeea03165a338d758f01ed

iandobbie, 2019-05-09:
Seems like a no brainer. The current dialog is almost useless, very difficult to read, and not being able to copy and paste is a continual annoyance to me.

carandraug, 2019-05-10:
I have proposed two different versions, it's not clear which one you guys prefer.

I will push the one without collapsible pane. Its code is simpler and I've found that the traceback is often needed to interpret the error message. Might as well have it displayed by default. A collapsible pane only makes sense if by default is collapsed.

carandraug, 2019-05-10:
Pushed the changes. I have pushed two commits so that if we ever want to have the traceback by default, it's already in the git history.

closed 2019-05-10 by carandraug