microscope-cockpit/cockpit #469: right click on laser 'always on' mode leaves laser on

juliomateoslangerak, 2019-04-12:
Right click on laser button turns laser 'always on' (no digital modulation). To exit that state:
- Left click turns 'always on; state off and goes into digital modulation mode. That behaviour is ok.
- Right click turns shows 'always on' mode off and digital modulation mode (green button) but laser stays on.
Can you verify if this is the case with your systems?

mickp, 2019-04-12:
New UI changes that use native elements will remove that capability
altogether. The only time we ever found a use for it was during alignment,
and for that case the same can be achieved by setting the appropriate line
high on the trigger source via its debug window.

juliomateoslangerak, 2019-05-07:
I guess this may be closed now

closed 2019-05-07 by juliomateoslangerak