microscope-cockpit/cockpit #476: Mac opengl issue.

iandobbie, 2019-04-30:
There is a new issue with opengl on Mac. Cockpit previously used to work but now crashes when it tries to draw a image to the camera canvas with...

Error drawing view canvas: b”ERROR: 0:2: ‘’ :  version ‘130’ is not supported\n”
Error drawing view canvas: GLError(
	err = 1282,
	description = b’invalid operation’,
	baseOperation = glUseProgram,
	cArguments = (1,)
)

It works fine on mickp/uiRework which was the last version I ran on this machine.

mickp, 2019-04-30:
mickp/uiRework inlcuded the glUpdates that cause this. How long ago did you
last fetch it?
I think this is probably to do with pixel formats. The old code used
GL_LUMINANCE which was deprecated, probably prior to GL 3.0, which means no
glsl 1.30 on the Mac. It'll probably work if we change the texture format
to GL_RED.

iandobbie, 2019-04-30:
Don't know exactly when I last pulled it . I am just trawling the diff to see what might have trigger it. I will try to pull the latest uiRework and see if that is broken or not.

Latest git log is 

commit 1542d98db0eaaf760472ae483c24e780bc5a7d0e (HEAD -> uiRework, mickp/uiRework)
Author: Mick Phillips <mick.phillips@gmail.com>
Date:   Mon Apr 8 10:54:05 2019 +0100

    Use constant instead of literal; remove unused line.

mickp, 2019-04-30:
Try mickp/GL_RED. The images will probably be red. If it works, I'll look
at fixing the grayscale.

mickp, 2019-04-30:
mickp/uiRework doesn't exist any more, because it was merged.

iandobbie, 2019-04-30:
Nope still broken with same Error message. I'm off now but will have more of a look later.

mickp, 2019-04-30:
1542d98 is from 3 weeks ago (became 6e66a07 when rebased and merged).

The Mac GL implementation forces GL2.1 (and glsl 1.20) unless you jump through some hoops. The new shaders need at least gl 1.30. We need to convince the Mac to give us Gl 3.0 Core or later.

mickp, 2019-05-01:
OK - fixed, and will PR in a bit.

mickp, 2019-05-01:
Closed by #477.

closed 2019-05-01 by mickp