microscope-cockpit/cockpit #478: Exception in executing experiments

NickHallONBI, 2019-05-01:
Tried to run a simple z-stack using the Micron Oxford "master" branch and got the following error:

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 207, in executeTable
    len(actions), numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 323, in executeTable
    lowestAnalogs = [min(channel) for channel in zip(*zip(*zip(*actions)[1])[1])]
TypeError: 'zip' object is not subscriptable

Didn't encounter this error before the Python3 update and this is the first time I've tried to run any since the update.

NickHallONBI, 2019-05-01:
These are the "actions" described in the penultimate line:

[(0.0, (513, [32636.928, 32768, 32768, 32768])), (0.1, (1, [32636.928, 32768, 32768, 32768])), (100.0, (0, [32636.928, 32768, 32768, 32768])), (100.0000000001, (0, [32636.928, 32768, 32768, 32768])), (100.0500000001, (0, [32663.1424, 32768, 32768, 32768])), (221.24959189891814, (513, [32663.1424, 32768, 32768, 32768])), (221.34959189891816, (1, [32663.1424, 32768, 32768, 32768])), (321.24959189891814, (0, [32663.1424, 32768, 32768, 32768])), (321.2495918990181, (0, [32663.1424, 32768, 32768, 32768])), (321.29959189901814, (0, [32689.3568, 32768, 32768, 32768])), (442.4991837978363, (513, [32689.3568, 32768, 32768, 32768])), (442.5991837978363, (1, [32689.3568, 32768, 32768, 32768])), (542.4991837978363, (0, [32689.3568, 32768, 32768, 32768])), (542.4991837979363, (0, [32689.3568, 32768, 32768, 32768])), (542.5491837979363, (0, [32715.5712, 32768, 32768, 32768])), (663.7487756967545, (513, [32715.5712, 32768, 32768, 32768])), (663.8487756967545, (1, [32715.5712, 32768, 32768, 32768])), (763.7487756967545, (0, [32715.5712, 32768, 32768, 32768])), (763.7487756968544, (0, [32715.5712, 32768, 32768, 32768])), (763.7987756968545, (0, [32741.785600000003, 32768, 32768, 32768])), (884.9983675956726, (513, [32741.785600000003, 32768, 32768, 32768])), (885.0983675956726, (1, [32741.785600000003, 32768, 32768, 32768])), (984.9983675956726, (0, [32741.785600000003, 32768, 32768, 32768])), (984.9983675957726, (0, [32741.785600000003, 32768, 32768, 32768])), (985.0483675957726, (0, [32768.0, 32768, 32768, 32768])), (1106.2479594945908, (513, [32768.0, 32768, 32768, 32768])), (1106.3479594945907, (1, [32768.0, 32768, 32768, 32768])), (1206.2479594945908, (0, [32768.0, 32768, 32768, 32768])), (1206.2479594946908, (0, [32768.0, 32768, 32768, 32768])), (1206.2979594946908, (0, [32794.2144, 32768, 32768, 32768])), (1327.497551393509, (513, [32794.2144, 32768, 32768, 32768])), (1327.5975513935089, (1, [32794.2144, 32768, 32768, 32768])), (1427.497551393509, (0, [32794.2144, 32768, 32768, 32768])), (1427.497551393609, (0, [32794.2144, 32768, 32768, 32768])), (1427.547551393609, (0, [32820.4288, 32768, 32768, 32768])), (1548.7471432924272, (513, [32820.4288, 32768, 32768, 32768])), (1548.847143292427, (1, [32820.4288, 32768, 32768, 32768])), (1648.7471432924272, (0, [32820.4288, 32768, 32768, 32768])), (1648.747143292527, (0, [32820.4288, 32768, 32768, 32768])), (1648.7971432925272, (0, [32846.6432, 32768, 32768, 32768])), (1769.9967351913451, (513, [32846.6432, 32768, 32768, 32768])), (1770.0967351913453, (1, [32846.6432, 32768, 32768, 32768])), (1869.9967351913451, (0, [32846.6432, 32768, 32768, 32768])), (1869.9967351914452, (0, [32846.6432, 32768, 32768, 32768])), (1870.0467351914451, (0, [32872.8576, 32768, 32768, 32768])), (1991.2463270902633, (513, [32872.8576, 32768, 32768, 32768])), (1991.3463270902635, (1, [32872.8576, 32768, 32768, 32768])), (2091.2463270902635, (0, [32872.8576, 32768, 32768, 32768])), (2091.2463270903636, (0, [32872.8576, 32768, 32768, 32768])), (2091.2963270903633, (0, [32899.072, 32768, 32768, 32768])), (2212.4959189891815, (513, [32899.072, 32768, 32768, 32768])), (2212.5959189891814, (1, [32899.072, 32768, 32768, 32768])), (2312.4959189891815, (0, [32899.072, 32768, 32768, 32768])), (2312.4959189892816, (0, [32899.072, 32768, 32768, 32768])), (2312.5459189892817, (0, [32636.928, 32768, 32768, 32768])), (2312.5959189892815, (0, [32636.928, 32768, 32768, 32768]))]

iandobbie, 2019-05-01:
This has all the hallmarks of a python3 ism!

NickHallONBI, 2019-05-01:
Tried David's branch "issue478" and got a new error:

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 207, in executeTable
    len(actions), numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 366, in executeTable
    maxticks = reduce(max, chain(zip(*digitals)[0],
NameError: name 'reduce' is not defined

NickHallONBI, 2019-05-01:
This error with "reduce" does appear to exist in the master branch, but at line 379 instead of 366. However, because the original error occurs at line 323, it is never seen

carandraug, 2019-05-01:
That's another python 2 -> 3 issue. reduce is now in the functools module.

NickHallONBI, 2019-05-01:
Adding functools still yields the following error:

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 207, in executeTable
    len(actions), numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 368, in executeTable
    maxticks = functools.reduce(max, chain(zip(*digitals)[0],
TypeError: 'zip' object is not subscriptable

In addition, printed "analogs" and "digitals" which are:

analogs: [[(0, 0.0), (1001, 26.21440000000075), (3213, 52.4288000000015), (5425, 78.64319999999861), (7638, 104.857600000003), (9850, 131.07200000000012), (12063, 157.28639999999723), (14275, 183.50080000000162), (16488, 209.71519999999873), (18700, 235.92960000000312), (20913, 262.14400000000023), (23125, 0.0)], [(0, 0)], [(0, 0)], [(0, 0)]]

digitals: [(0, 513), (1, 1), (1000, 0), (1001, 0), (2212, 513), (2213, 1), (3212, 0), (3213, 0), (4425, 513), (4426, 1), (5425, 0), (6637, 513), (6638, 1), (7637, 0), (7638, 0), (8850, 513), (8851, 1), (9850, 0), (11062, 513), (11063, 1), (12062, 0), (12063, 0), (13275, 513), (13276, 1), (14275, 0), (15487, 513), (15488, 1), (16487, 0), (16488, 0), (17700, 513), (17701, 1), (18700, 0), (19912, 513), (19913, 1), (20912, 0), (20913, 0), (22125, 513), (22126, 1), (23125, 0), (23126, 0)]

juliomateoslangerak, 2019-05-01:
I guess you have to convert that into a list. In python 2 zip returns a list while in 3 is an iterable.

list(zip(...))

J

carandraug, 2019-05-01:
Proposed fix in my branch `fix-issue478` or carandraug/cockpit@529ecb0fec1acc57a72fe74e25dd41143e0143e8

NickHallONBI, 2019-05-02:
This now yields a new error:

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 207, in executeTable
    len(actions), numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 312, in executeTable
    if lowest < self._lastAnalogs[line]:
IndexError: list index out of range

I printed out "line", "lowest" and "self._lastAnalogs[line]" which yields the following:

line:  0
lowest:  32636.928
self._lastAnalogs[line]:  32636.928
line:  1
lowest:  32636.928
self._lastAnalogs[line]:  32768
line:  2
lowest:  32636.928
self._lastAnalogs[line]:  32768
line:  3
lowest:  32636.928
self._lastAnalogs[line]:  32768
line:  4
lowest:  32663.1424

mickp, 2019-05-02:
There aren't 5 analogue lines on the DSP, so there should only be lines 0,
1, 2 and 3. Check the config file for something being assigned to analogue
line 4.

NickHallONBI, 2019-05-02:
depot.conf from DeepSIM as per Mick's request:

[server]
ipAddress: 192.168.1.1

[dsp]
type: cockpit.devices.executorDevices.LegacyDSP
uri: PYRO:pyroDSP@192.168.1.4:8000

[SI polarizer]
type: cockpit.devices.polarizationRotator.PolarizationDevice
analogSource: dsp
analogLine: 1
#gain: 3.815844361
gain: 6553.6
offset: 0.947
siVoltages: 488: 0.915, 1.05, 1.23
            561: 1.32, 0.90, 1.12
            647: 1.18, 1.61, 0.97
idleVoltage: 1.0

[West-Red]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:AndorAtmcd@192.168.1.2:7777
triggerSource: dsp
triggerLine: 8
transform: (0,0,1)
wavelength: 610
isWaterCooled: True
targetTemperature: -80

[East-Green]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:AndorAtmcd@192.168.1.2:7778
triggerSource: dsp
triggerLine: 9
transform: (0,0,1)
wavelength: 525
isWaterCooled: True
targetTemperature: -80

[interference]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:XimeaCamera@192.168.1.20:8008
triggerSource: dsp
triggerLine: 1
transform: (0,0,1)

[405]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
uri: PYRO:DeepstarLaser@192.168.1.2:7770
wavelength: 406
triggerSource: dsp
triggerLine: 1

[488]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
uri: PYRO:DeepstarLaser@192.168.1.2:7771
wavelength: 489
triggerSource: dsp
triggerLine: 0

[561]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
uri: PYRO:CoboltLaser@192.168.1.2:7773
wavelength: 562
triggerSource: dsp
triggerLine: 3

[647]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
uri: PYRO:DeepstarLaser@192.168.1.2:7772
wavelength: 648
triggerSource: dsp
triggerLine: 2

[trans]
type: cockpit.devices.light.SimpleLight
triggerSource: dsp
triggerLine: 4

[m687]
type: cockpit.devices.physikInstrumenteM687.PhysikInstrumenteM687
port: com6
baud: 115200
timeout: 0.1
softlimits: ((-37500,-67500),(11500,59500))
primitives:  r, 500, -7000 ,22000,76000
             c, 6500, -6200, 4500

[zPiezo]
type: cockpit.devices.stage.SimplePiezo
analogSource: dsp
analogLine: 0
offset: 0
#gain: 0.1547362671
gain: 262.144
min: 0
range: 250

[aerotech]
type: cockpit.devices.aerotechAxis.AerotechZStage
ipAddress: 192.168.1.16
port: 8000
softlimits: -45000, 5000

[slm]
type: cockpit.devices.boulderSLM.BoulderSLM
triggerSource: dsp
triggerLine: 7
uri: PYRO:pyroSLM@192.168.1.19:8000
settlingtime: 10

[561 shutter]
type: cockpit.devices.sr470.StanfordShutter
ipAddress: 192.168.1.18
port: 5024
lights: 561
triggerSource: dsp
triggerLine: 3

[objectives]
10x: 0.792, 10118, (0, 1, 1), (-35004,-598,10000),(1,0,0)
60xWater: .1333,10618,(0, 1, 1),(0,0,0),(0,0,1)

[PiDIO]
type: cockpit.devices.piDIO.RaspberryPi
ipAddress: 192.168.1.10
port: 7768
lines: 'Objective', 'trans-light', 'WF-in', 'WF-out', 'DM bypass in', 'DM bypass out'
paths: objective:10x:0,1|1,0;objective:60xwater:0,0|1,1;excitation:conventional:2,1|3,1|4,0;excitation:SI:2,0|3,0|4,0|5,0;excitation:DM bypass:4,1|5,1

mickp, 2019-05-02:
Hmm.  Not that, then. I don't see the DM device in there. Are you using any
auxiliary config for that?

NickHallONBI, 2019-05-02:
Yes. The above config is the once stored in "C:\ProgramData\cockpit\depot.conf" which is the base config for all users. There is my user specific config which contains just the DM and nothing else. I've also commented it out for the moment as both the current master branch and the branches with the proposed fixes don't have a cockpit module to talk to the DM.

mickp, 2019-05-02:
Ah, right - I see it now. It's a py3 thing. Fixing now.

mickp, 2019-05-02:
carandraug/cockpit@529ecb0 replaces
```lowestAnalogs = [min(channel) for channel in zip(*zip(*zip(*actions)[1])[1])]```
with 
```lowestAnalogs = list(np.amin([x[1][1] for x in actions], axis=1))```
These don't do the same thing. The original gives a list with length = number of channels, having extracted the lowest analogue value used on each channel. The replacement gives a list with length = len(actions).

mickp, 2019-05-02:
It should be:
```lowestAnalogs = list(np.amin([x[1][1] for x in actions], axis=0))```

closed 2019-05-02 by mickp

NickHallONBI, 2019-05-02:
Experiments seem to run fine now

PR #479 (2019-05-02): Fix issue478
Fixes #478