microscope-cockpit/cockpit #480: Changing step size no longer works

NickHallONBI, 2019-05-02:
The keyboard shortcuts to increase/decrease the step sizes for both the coarse and fine z movers have now died. There is still keyboard control over switching between axis movers and moving the stage up and down, but the step sizes are locked to 50um and 0.25um respectively. There is no accompanying error.

Possible related to Issue #413 (as this also dealt with the step sizes) and/or Issue #478 as controlling the step size worked before those changes were made.

mickp, 2019-05-02:
NumPad PgUp and PgDn change the step size correctly for my test setup,
which uses a SimplePiezo as the Z device.
Which movers is it not working for? Piezo, Aerotech, or both?

NickHallONBI, 2019-05-02:
Both, however I have now realised that this is only an issue if the user doesn't have NumLock active. Activating this fixes the issue. If this is the intended behavior, then this is a non-issue and I apologize.

carandraug, 2019-05-02:
> Both, however I have now realised that this is only an issue if the user doesn't have NumLock active. Activating this fixes the issue. If this is the intended behavior

NumLock changes what key code you get, just like CapsLock and FnLock does. Let's not work around that. I remap some of the keys on my system and would not like a program trying to be smarter than me and remapping back to whatever the program thinks should be.

closed 2019-05-02 by carandraug