microscope-cockpit/cockpit #481: Zyla 'buffer_overflow_event' property assigned (None, None) values.

juliomateoslangerak, 2019-05-05:
With the new GUI rework clicking on my Zyla settings button pops an error in gui/device:441 when comparing two none objects. basically the 'buffer_overflow_event' property is assigned as an int but has a description of a two Nones tuple. 

I will have a further look into this and see where this comes from

iandobbie, 2019-05-05:
I would t be surprised if this is a python2-3 problem. I don’t think the Zyla driver has been tested in python3

Ian

juliomateoslangerak, 2019-05-07:
Dont think so. I alway srun the zyla with python 3.

mickp, 2019-05-07:
For an int, the description shows the min and max values. I would suspect that either the camera or the SDK is not returning min and max values for BufferOverflowEvent. You can temporarily disable it by commenting out (or near) line 134 of microscope.devices.cameras.andorsdk3. If you do this, do other ints work correctly?

juliomateoslangerak, 2019-05-07:
Yes, that is normal behavior. But indeed there are some settings where the camera or sdk returns None. They are events: see below. 
I guess, as you say, that they are not returned by camera or sdk. Is it a new sdk? I don't recall these settings from reading the sdk manual.

Commenting those settings out and no error anymore.

'events_missed_event' = {dict} {'type': 'int', 'values': (None, None), 'readonly': True, 'cached': False}
'exposure_end_event' = {dict} {'type': 'int', 'values': (None, None), 'readonly': True, 'cached': False}
'exposure_start_event' = {dict} {'type': 'int', 'values': (None, None), 'readonly': True, 'cached': False}
'row_n_exposure_end_event' = {dict} {'type': 'int', 'values': (None, None), 'readonly': True, 'cached': False}
'row_n_exposure_start_event' = {dict} {'type': 'int', 'values': (None, None), 'readonly': True, 'cached': False}

mickp, 2019-05-08:
The microscope Setting docstring implies that 'values' is only used to determine what the allowable values are for setting values (but we also use it to convert enum indices to values). For read-only ints, ```(None, None)``` is an acceptable state for ```values```.
The problem line in cockpit is a fudge because python ```ints``` can be larger than wx's IntProperty, depending on a compile-time flag, ```wxUSE_LONG_LONG```.  I check the max. value to see if it exceeds ```wx.INT32_MAX``` and, if it does, use a FloatProperty.

One fix might be to use FloatProperties for read-only ints, because we have no idea what their max. value might be. This works fine for permanently read-only Settings, because the control is never used to modify the Setting, so floating point error is not an issue. However, there are some Settings whose read-only state changes depending on the device state.

PR #488 (2019-05-08): Avoid max() comparison on (None, None).
Fixes #481.
closed 2019-05-08 by mickp