microscope-cockpit/cockpit #482: microscope camera 'enum' properties not matching

juliomateoslangerak, 2019-05-07:
I don''t know if this might belong in microscope.

I merged to master branches of cockpit and microscope and there is a mismatch between the communication of the 'enum' settings.
So when I click on the settings of the camera enum settings appear in red as if the settings are not in the available options. It seems that cockpit likes for the enum settings description values a list of index-string and that microscope camera has implemented these as a list of strings.

>self.proxy.update_settings({'aoi_binning':'4x4'})
>self.proxy.get_setting('aoi_binning')
'4x4'
 But cockpit is trying to change using the index integer.

`Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\device.py", line 338, in <lambda>
    applyButton.Bind(wx.EVT_BUTTON, lambda evt: self.device.updateSettings(self.current))
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\interfaces\imager.py", line 79, in wrapper
    result = func(*args, **kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\devices\microscopeCamera.py", line 112, in updateSettings
    self.proxy.update_settings(settings)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
ctypes.ArgumentError: argument 3: <class 'TypeError'>: wrong type
`

mickp, 2019-05-07:
Yes - this was changed in the UI rework in cockpit with corresponding changes in microscope to avoid potential errors when passes enums back and forth.

juliomateoslangerak, 2019-05-07:
Still, in my setup there is problem. I verified that I merged microscope and cockpit masters.
There is a different behavoiur between microscope generated enum settings and sdk generated enum settings. Or maybe cause the values are tuples and not strings

>self.proxy.describe_setting('transform')
{'type': 'enum', 'values': [(0, (0, 0, 0)), (1, (0, 0, 1)), (2, (0, 1, 0)), (3, (0, 1, 1)), (4, (1, 0, 0)), (5, (1, 0, 1)), (6, (1, 1, 0)), (7, (1, 1, 1))], 'readonly': False, 'cached': False}
>self.proxy.get_setting('transform')
7
>self.proxy.set_setting('transform', 5)
>self.proxy.get_setting('transform')
5

However:
>self.proxy.describe_setting('aoi_binning')
{'type': 'enum', 'values': [(0, '1x1'), (1, '2x2'), (2, '3x3'), (3, '4x4'), (4, '8x8')], 'readonly': False, 'cached': False}
> self.proxy.get_setting('aoi_binning')
'1x1'
> self.proxy.set_setting('aoi_binning', 1)  # Using the index is how cockpit changes settings through updateSettings and where things go wrong
> self.proxy.get_setting('aoi_binning')
'1x1'
> self.proxy.set_setting('aoi_binning', '2x2')
> self.proxy.get_setting('aoi_binning')
'2x2'

juliomateoslangerak, 2019-05-07:
Replacing in lines 348/349 of micrsocope andorSDK for this solves the problem. Can yo uverify this is ok?

                    set_func = var.set_index
                    get_func = var.get_index

mickp, 2019-05-08:
I've looked through the cockpit-side and microscope-side code for settings. The behaviour for enum-type settings hasn't been written down anywhere. The key question is what property of the setting should be returned/passed by get_value and set_value: should it be the index of the required value, the value itself, or a combination? There are pros and cons to each.

The value of an enum may be of any type -  a string, a tuple, or something more complex, exposed as a string-like representation in cockpit. I decided to use indices to avoid problems passing arbitrary types back and forth over Pyro, and errors in recreating an object from its string-like representation.

On the other hand, there's nothing to say that the possible values of an enum setting are immutable: if they change, then that means the indices on the client (cockpit) and device might get out of sync. (So far, however, I'm not aware of any enum settings that change like this.)

I see four possible options:
  i) set and get enum values;
  ii) set and get indices;
  iii) set and get tuples of index and value.__repr__().
  iv) get tuples, but allow either ii) or iii) for set.

I think (i) is undesirable - there's too much scope for errors during conversion/serialisation/deserialisation.

(ii) is simple - it's what the test devices already do.

(iii) allows us to deal with mutable possible-values - on set(index, repr), we can compared repr to possible_values[index].__repr__, and throw an exception or warn that the setting has mutated.

(iv) allows us to deal with mutable possible-values, or just accept indices where the possible values never change.

Is it worth worrying about this, given that nothing we've yet seen has varying possible-values, or do we just go with (ii)?

mickp, 2019-05-08:
OK - this should be fixed by MicronOxford/microscope#82

closed 2019-05-08 by mickp

carandraug, 2019-05-10:
> The behaviour for enum-type settings hasn't been written down anywhere. 

I had a related issue about this some time ago.  See MicronOxford/microscope#70 where I argue that `Setting.get` and `Setting.set` should be returning/accepting the actual enum instances.