microscope-cockpit/cockpit #484: Andor EMCCD cameras in microscope no longer return temperature.

NickHallONBI, 2019-05-07:
The microscopeversion of the andor ixon driver doesnt give a log of temperature any more so cannot be followed or checked.

mickp, 2019-05-07:
Yep. They're now implementations of  microscope CameraDevices. Temperature
info isn't part of the CameraDevice interface, so we probably need to sort
out how to deal with composite devices to bring the temperature logging
back.

mickp, 2019-06-13:
I added temperature to AndorAtmcd's settings in MicronOxford/microscope@4ebf36da23fd9276e25658a97651ef2fad3aa3a5 - the current temperature will now show up as a read-only value in the cameras settings in cockpit. We don't log it, though, and I don't think settings is the right thing to use for logging - we should probably use methods on the remote to access those values. We have talked briefly about a Sensors mix-in in microscope that adds these methods, so we should explore that.

iandobbie, 2023-03-16:
There is now a ValueLogger device type, but maybe we also need a Value Logger mixin.

By having a mixin and some mechanism to have both push and pull Value logger devices might solve a few comments that have come up in the recent discussion such as #855