microscope-cockpit/cockpit #485: More python2-3 issues in experiments.

NickHallONBI, 2019-05-07:
Running an SI experiment generates an error....

Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\experiment\experiment.py", line 397, in execute
    curIndex + bestLen, numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\events.py", line 187, in executeAndWaitFor
    func(*args, **kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\handlers\executor.py", line 207, in executeTable
    len(actions), numReps, repDuration)
  File "c:\cockpit\nick\cockpit\cockpit\devices\executorDevices.py", line 310, in executeTable
    if lowest < self._lastAnalogs[line]:
TypeError: 'map' object is not subscriptable

Obviously this is due to the change in the map behavior between python2 and 3. We ought to test all the experiments ASAP.

NickHallONBI, 2019-05-07:
Quick tests show Z-stack, rotator sweep work but 3D and 2D SIM experiments both fail with this error.

mickp, 2019-05-07:
Dammit. Some map needs wrapping with a list().

mickp, 2019-05-08:
Should be fixed by #486. Please test and comment or close.

NickHallONBI, 2019-05-08:
This fix works, cheers!

closed 2019-05-08 by NickHallONBI