microscope-cockpit/cockpit #493: Mosaic deals fine with rectangular images but the viewcanvas doesn't

iandobbie, 2019-05-09:
The picam returns at 720x480 image. Teh mosaic tiles these in the correct place and seems to move the right amount (in simulated stage mode anyway). However, the main camera view window just doesn't cope with rectangular image and squashes it to square on the display

iandobbie, 2019-05-09:
I looked at the code but it is pretty deep GL stuff. I think it is mapping the pixels over the square viewport, no matter hjow many their are in each direction, but I didn't get to the bottom of it.

carandraug, 2019-05-15:
I'm having the same issue now with the IDS cameras. The sensor size is rectangular (1936x1216), the camera view is square, so the pixels on the camera view end up rectangular.

mickp, 2019-05-16:
I fixed this on the camera by adjusting the quad co-ordinates. While mosaic looks like it's dealing with non-square images, I think it's doing it incorrectly: it looks like it's scaling the pixels along the wrong axis.
[image]

mickp, 2019-05-21:
#499 will fix this.

iandobbie, 2020-04-28:
This is fixed so closing

closed 2020-04-28 by iandobbie