microscope-cockpit/cockpit #497: New traceback windows too narrow under MSW

mickp, 2019-05-14:
[image]

carandraug, 2019-05-14:
The issue here is not windows, the issue is when the error message is too short. What happens is that the default size of a TextCtrl (where the traceback is) does not make use of its actual content so the size of the StaticText where the error message is is used. 

I have now found a way to get the size required for the traceback but that solution does not work in Windows. I will look deeper tomorrow.

carandraug, 2019-05-15:
I found a solution that works fine. Is not perfect though. On windows it fails to estimate the correct height of the traceback but not by much, which I think is good enough.

closed 2019-05-15 by carandraug

mickp, 2019-05-15:
If the width is set by the width of the exception message, perhaps you could just pad the message with whitespace, if that avoids the problem with traceback height estimation?

carandraug, 2019-05-15:
> If the width is set by the width of the exception message, perhaps you could just pad the message with whitespace, if that avoids the problem with traceback height estimation?

The height is now computed correctly even on windows. I split the traceback by lines and used the width needed for the longest line. The problem now is the height on windows. It's not off by much anyway.