microscope-cockpit/cockpit #503: "Warn if MicroscopeCamera returns bare tuples" issue

NickHallONBI, 2019-06-05:
There seems to be an issue with commit 8527b3b242b146a41cebdcca21dd908c5dca7861. It imports ROI and Binning from microscope.devices but the current version of microscope from PyPi doesn't have those and so the initialisation fails. Not so much of an issue for people who will just clone the current microscope repo where those things can be imported from, but it will fail for someone who just does "pip install microscope" and expects cockpit to work

carandraug, 2019-06-05:
If you are using microscope from PyPI with current cockpit, the missing ROI and Binning class is not your only problem. I remember being some other things that were mixed with adding support for the Andor EMCCD.

mickp, 2019-06-05:
The current version of Microscope from PyPi is a year old, so is missing a
whole bunch of stuff. I think we decided not to roll releases until we had
something stable worth releasing.

iandobbie, 2019-06-05:
Yes we should think about rolling a release with the python 3 support done and a bunch of other changes. Happy to discus about when but we should do it quite soon

Ian

mickp, 2019-06-13:
Closing this - it was solely due to using an old version.

closed 2019-06-13 by mickp